ARISE-Initiative/robosuite #555: Viewer freezes after second reset

nickstr15, 2024-11-14:
### System Info

```Shell
- Ubuntu 24.04.
- robosuite v1.5
```

### Information

Running multiple episodes results in the viewer freezing. 

The simulation is running correctly (as can be seen in the plot):

[image]

But the viewer window freezes after the first episode:

https://github.com/user-attachments/assets/67a534a9-eae1-45b6-9d95-8d370adbd8a5

Console:
```
[robosuite WARNING] Could not load the mink-based whole-body IK. Make sure you install related import properly, otherwise you will not be able to use the default IK controller setting for GR1 robot. (__init__.py:40)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)
/home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/glfw/__init__.py:914: GLFWError: (65544) b'Wayland: Window position retrieval not supported'
  warnings.warn(message, GLFWError)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)
[robosuite INFO] Loading controller configuration from: /home/nick/miniconda3/envs/bil-py310/lib/python3.10/site-packages/robosuite/controllers/config/robots/default_panda.json (composite_controller_factory.py:69)

Process finished with exit code 0
```

### Reproduction

Run the following code to reproduce:

```python
import numpy as np
import matplotlib.pyplot as plt
import robosuite as suite

def reset_loop():
    env = suite.make(
        "TwoArmHandover",
        robots=["Panda", "Panda"],
        env_configuration="parallel",
        has_renderer=True,
        render_camera="frontview",
        camera_names="frontview",
        has_offscreen_renderer=True,
        use_object_obs=True,
        use_camera_obs=True,
    )

    env.deterministic_reset = False

    hammer_positions_after_reset = []
    imgs_after_reset = []
    for i in range(4):
        obs = env.reset()
        img = obs["frontview_image"]
        hammer_pos = obs["hammer_pos"]

        hammer_positions_after_reset.append(hammer_pos)
        imgs_after_reset.append(img)

        # some dummy steps
        for _ in range(100):
            action = np.random.uniform(env.action_spec[0])*0.1
            _ = env.step(action)

    fig, axs = plt.subplots(2, 2, figsize=(8, 8), tight_layout=True)
    # print the hammer positions
    for i, (pos, img, ax) in enumerate(
            zip(hammer_positions_after_reset, imgs_after_reset, axs.flatten())
    ):
        ax.set_title(f"Hammer position after reset {i}\n{pos}")
        ax.imshow(img, origin="lower")

    plt.show()

if __name__ == "__main__":
    reset_loop()
```

### Expected behavior

The viewer should not freeze.

nickstr15, 2024-11-14:
I could fix this by adding 
```python
env.viewer.close()
env.viewer = None
```
after each episode, but I think this is not the expected behavior.

kevin-thankyou-lin, 2024-11-14:
Thanks @nickstr15! PR coming soon

closed 2024-12-16 by kevin-thankyou-lin