ARISE-Initiative/robosuite #559: installation error(about evdev)

liaojiaxin97, 2024-11-15:
### System Info

```Shell
Robosuite 1.5.0
Ubuntu 11.4.0

pip list
absl-py                 2.1.0
aiohappyeyeballs        2.4.0
aiohttp                 3.10.5
aiosignal               1.2.0
antlr4-python3-runtime  4.9.3
asciitree               0.3.3
async-timeout           4.0.3
attrs                   24.2.0
av                      13.1.0
Bottleneck              1.3.7
Brotli                  1.0.9
certifi                 2024.8.30
cffi                    1.17.1
charset-normalizer      3.3.2
CleanDiffuser           0.1.0       /home/pxw/QDU/Neo/Project/CleanDiffuser-main/CleanDiffuser-main
click                   8.1.7
cloudpickle             3.1.0
contourpy               1.2.1
cycler                  0.12.1
Cython                  3.0.0a10
D4RL                    1.1         /home/pxw/QDU/Neo/Project/CleanDiffuser-main/D4RL-master
datasets                2.12.0
dill                    0.3.6
dm-control              1.0.16
dm-env                  1.6
dm-tree                 0.1.8
docker-pycreds          0.4.0
egl_probe               1.0.2
einops                  0.8.0
etils                   1.5.2
Farama-Notifications    0.0.4
fasteners               0.19
filelock                3.13.1
fonttools               4.54.1
frozenlist              1.5.0
fsspec                  2024.10.0
gitdb                   4.0.11
GitPython               3.1.43
glfw                    2.7.0
gmpy2                   2.1.2
grpcio                  1.67.1
gym                     0.23.1
gym-notices             0.0.8
gymnasium               1.0.0
gymnasium-robotics      1.3.1
h5py                    3.12.1
huggingface_hub         0.24.6
hydra-core              1.3.2
idna                    3.7
imagecodecs             2024.9.22
imageio                 2.36.0
imageio-ffmpeg          0.5.1
importlib_metadata      8.5.0
importlib_resources     6.4.5
Jinja2                  3.1.4
kiwisolver              1.4.7
labmaze                 1.0.6
lazy_loader             0.4
llvmlite                0.42.0
lxml                    5.3.0
Markdown                3.7
MarkupSafe              2.1.3
matplotlib              3.7.5
mjrl                    1.0.0
mkl-fft                 1.3.1
mkl-random              1.2.2
mkl-service             2.4.0
mpmath                  1.3.0
mujoco                  3.1.6
mujoco-py               2.1.2.14
multidict               6.1.0
multiprocess            0.70.14
networkx                3.2.1
numba                   0.59.1
numcodecs               0.12.1
numexpr                 2.8.4
numpy                   1.22.4
omegaconf               2.3.0
opencv-python           4.10.0.84
packaging               24.1
pandas                  2.2.2
pettingzoo              1.24.3
pillow                  10.4.0
pip                     24.2
platformdirs            4.3.6
protobuf                5.28.3
psutil                  6.1.0
pyarrow                 11.0.0
pybullet                3.2.6
pycparser               2.22
pygame                  2.6.1
pymunk                  6.9.0
PyOpenGL                3.1.7
pyparsing               3.2.0
PySocks                 1.7.1
python-dateutil         2.9.0.post0
pytz                    2024.1
PyYAML                  6.0.2
requests                2.32.3
responses               0.13.3
robomimic               0.3.0
scikit-image            0.22.0
scipy                   1.13.1
sentry-sdk              2.18.0
setproctitle            1.3.3
setuptools              75.1.0
Shapely                 1.8.5.post1
six                     1.16.0
smmap                   5.0.1
sympy                   1.13.2
tensorboard             2.18.0
tensorboard-data-server 0.7.2
tensorboardX            2.6.2.2
termcolor               2.5.0
tifffile                2024.8.30
torch                   2.2.2
torchvision             0.17.2
tqdm                    4.66.5
triton                  2.2.0
typing_extensions       4.11.0
tzdata                  2023.3
urllib3                 2.2.3
wandb                   0.18.5
Werkzeug                3.1.3
wheel                   0.44.0
xxhash                  2.0.2
yarl                    1.11.0
zarr                    2.16.1
zipp                    3.20.2
```

### Information

The problem arises when running"pip install robosuite-1.5.0-py3-none-any.whl",the error message is "  
note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for evdev
Failed to build evdev
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (evdev)
"

### Reproduction

None

### Expected behavior

try“sudo apt-get install python3-dev”，but make no sense

kevin-thankyou-lin, 2024-11-15:
Can you try running `pip install robosuite` instead of pip install robosuite-1.5.0-py3-none-any.whl?

liaojiaxin97, 2024-11-15:
> Can you try running `pip install robosuite` instead of pip install robosuite-1.5.0-py3-none-any.whl?
With the former method, the download speed is only 20kb

kevin-thankyou-lin, 2024-11-15:
```
Downloading robosuite-1.5.0-py3-none-any.whl (150.1 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 150.1/150.1 MB 52.2 MB/s eta 0:00:00
```

For me it's pretty fast. That said, have you tried chatgpt'ing the error?

[image]

If that doesn't work, perhaps starting from a clean conda/mamba env and trying `pip install robosuite`

closed 2024-11-22 by liaojiaxin97