ARISE-Initiative/robosuite #562: OSC controller with control_delta=False does not work in 1.5.0

YukiShirai, 2024-11-15:
### System Info

```Shell
Robosuite 1.5.0
Ubuntu 22.04
```

### Information

I have been trying to run OSC controllers with `control_delta=False` in robosuite in 1.5.0. However, it does not move as I expected. I attached two codes below.

While I debug robosuite, I noticed that [`input_type`](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/parts/arm/osc.py#L135)  is set to `delta` even with `control_delta=False`. Thus, the robot was running with delta motion while users provide position in base frame. 

Once I force to set `input_type=absolute`, now [this line ](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/parts/arm/osc.py#L263)complains. I think there is something wrong in action space if I force to set `input_type=absolute`. 

When I run the very similar implementation in robosuite 1.4.1, which is attached below, I observe the expected motion, meaning that the robot moves +0.1 m along all x, y, z axes from the initialized position.

### Reproduction

```
import robosuite as suite
import numpy as np
from robosuite.controllers.composite.composite_controller_factory import refactor_composite_controller_config

arm_controller_config = suite.load_part_controller_config(
    default_controller='OSC_POSITION',
)

options = {}

options['env_name'] = 'Door'
options['robots'] = 'UR5e'
robot = options['robots'][0] if isinstance(
    options['robots'], list,
) else options['robots']
options['initialization_noise'] = None
options['controller_configs'] = refactor_composite_controller_config(
    arm_controller_config, robot, ['right', 'left'],
)

arm_controller_config['control_delta'] = False

env = suite.make(
    **options,
    use_camera_obs=False,  # Set to True if you want camera observations
    has_offscreen_renderer=False,
    has_renderer=True,  # Enable on-screen rendering
    control_freq=200,  # Control frequency in Hz
)

obs = env.reset()
env.viewer.set_camera(camera_id=0)

target_move = 0.1  # 0.1 meters along x-axis
target = obs["robot0_eef_pos"] + np.array([target_move, target_move, target_move])

# Run a few steps to reach the target position
for i in range(1000):

    # Set desired action in absolute frame
    action = np.zeros(4)
    action[:3] = target

    # Send action to the environment
    obs, reward, done, info = env.step(action)

    # Render the environment for visualization
    env.render()

# Close the environment
env.close()
```

1. Please try to run the above code in robosuite 1.5.0. The robot keeps moving and never converges to its desired state, which is +0.1 m in x, y, z axis from the initialized position. 

```
import robosuite as suite
from robosuite.controllers import load_controller_config
import numpy as np

# Load controller configuration for OSC_POSITION
controller_config = load_controller_config(default_controller="OSC_POSITION")
controller_config['control_delta'] = False
# Create the environment with OSC_POSITION as the controller
env = suite.make(
    env_name="Lift",  # You can replace "Lift" with any other suitable task/environment
    robots="Panda",  # Choose your robot, e.g., Panda, Sawyer
    controller_configs=controller_config,
    use_camera_obs=False,  # Set to True if you want camera observations
    has_offscreen_renderer=False,
    has_renderer=True,  # Enable on-screen rendering
    control_freq=200,  # Control frequency in Hz
)

# Reset the environment
obs = env.reset()
env.viewer.set_camera(camera_id=0)  # Optional: Adjust the camera view if using GUI

# Define the movement in x-axis (positive or negative for direction)
target_x_move = 0.1  # 0.1 meters
target = obs["robot0_eef_pos"] + np.array([target_x_move, target_x_move, target_x_move])
# Run a few steps to reach the target position
for i in range(1000):  # Adjust range to control duration
    # Get the current position of the robot end-effector
    # current_pos = obs["robot0_eef_pos"]

    # Set desired action in absolute frame
    action = np.zeros(4)
    action[:3] = target

    # Send action to the environment
    obs, reward, done, info = env.step(action)

    # Render the environment for visualization
    env.render()

# Close the environment
env.close()
```

2. Please try to run the above code in robosuite 1.4.1. In this code, the robot could successfully move to its desired state. 

### Expected behavior

The robot end-effector should converge to 0.1 m in all axes from the initialized position using robosuite 1.5.0.

kevin-thankyou-lin, 2024-11-16:
Thanks for raising this issue @YukiShirai! For now, I'd recommend using `osc_pose` and setting a fixed rotation.

@snasiriany: could you look into this one? The new osc.py logic doesn't handle osc position correctly

snasiriany, 2024-12-16:
Hi @YukiShirai the control_delta feature has been deprecated. It's now replaced with `input_type`, which can be set to either "absolute" or "delta". For example taking your example code into account:
```
options['controller_configs']["body_parts"]["right"]['input_type'] = "absolute"
```

I see the error that's caused here, and it's specifically with OSC_POSITION. I'm looking into a solution and will propose a fix soon.

snasiriany, 2024-12-17:
@YukiShirai another point I forgot to mention: the default frame of reference for actions has been changed from world frame to base frame. To be backwards compatible using the example you shared, you need to set the reference frame accordingly as well:
```
options['controller_configs']["body_parts"]["right"]['input_ref_frame'] = "world"
```

closed 2024-12-17 by kevin-thankyou-lin

lorepieri8, 2025-05-29:
Notice that the docs still refers to control_delta: 
https://robosuite.ai/docs/modules/controllers.html#:~:text=control_delta%3A%20Only%20relevant%20for%20OSC_POSE%20or%20OSC_POSITION%20controllers.%20true%20interprets%20input%20actions%20as%20delta%20values%20from%20the%20current%20robot%20end%2Deffector%20position.%20Otherwise%2C%20assumed%20to%20be%20absolute%20(global)%20values