ARISE-Initiative/robosuite #563: Reward modification to stabilize robot

garg-akash, 2024-11-16:
Hello,

Firstly, thank you for providing such an interesting framework. 

As I trained a DDPG policy on a Universal Robots model for cube lifting task using dense reward, I noticed that although the robot can pick the cube but it's quite unstable at times (jerk, oscillations etc.). It is most likely due to the fact that the reward function only considers if the robot is approaching the cube if the cube has been lifted above the tabletop. 

Is it a good idea to formulate a custom reward function that penalizes, for example, large robot joint velocity changes? Am I missing something about the environment here?

Thank you for the support in advance!
Akash

kevin-thankyou-lin, 2024-11-16:
Sure! Which controller are you using --- is it the OSC controller? Yes - you can certainly try adding extra reward terms in a custom reward function.

garg-akash, 2024-11-17:
Thank you for the prompt reply.

Yes, I'm using OSC controller at the moment.

kevin-thankyou-lin, 2024-11-17:
Gotcha, yeah adding some extra reward term would make sense

closed 2024-11-27 by kevin-thankyou-lin