ARISE-Initiative/robosuite #573: New PR I made breaks rendering when vram is contested

mishmish66, 2024-11-21:
### System Info

```Shell
I am using Linux ubuntu 24.04 but I suspect this would affect every version of robosuite on all systems.
```

### Information

This problem arises when VRAM is contested and offscreen rendering is used.
I introduced this issue with PR #570

I'll introduce a fix for this momentarily. So sorry to introduce this but hopefully I can get this fix in fast.

### Reproduction

Run a large model and robosuite offscreen rendering at the same time

### Expected behavior

I would expect robosuite rendering to be agnostic to the amount of VRAM available and in use

mishmish66, 2024-11-21:
While this is still an issue I've realized this isn't due to the PR I made (whew) here is a video of what happens.

> [!WARNING]  
> Flashing colors
> https://github.com/user-attachments/assets/37368a53-4311-4ebe-85c8-05be796fcc1a

I'll work on figuring out what's going on here.

kevin-thankyou-lin, 2024-11-22:
Hey @mishmish66, thanks for the PR! Do you remember the command that causes this offscreen rendering issue?