ARISE-Initiative/robosuite #576: AttributeError: module 'robosuite' has no attribute 'make'

liaojiaxin97, 2024-11-22:
### System Info

```Shell
name: robosuite
Version: 1.5.0

Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
```

### Information

The problem arises when running test scripts
‘’‘’import numpy as np
import robosuite as suite

# create environment instance
env = suite.make(
    env_name="Lift", # try with other tasks like "Stack" and "Door"
    robots="Panda",  # try with other robots like "Sawyer" and "Jaco"
    has_renderer=True,
    has_offscreen_renderer=False,
    use_camera_obs=False,
)

# reset the environment
env.reset()
‘’‘’
for i in range(1000):
    action = np.random.randn(*env.action_spec[0].shape) * 0.1
    obs, reward, done, info = env.step(action)  # take action in the environment
    env.render()  # render on display
‘’‘’ in https://robosuite.ai/docs/basicusage.html

 the error message is 
'''
Traceback (most recent call last):
  File "/home/pxw/QDU/Neo/Project/CleanDiffuser-main/CleanDiffuser-main/test.py", line 51, in <module>
    env = suite.make(
AttributeError: module 'robosuite' has no attribute 'make'
'''

### Reproduction

None

### Expected behavior

What's the problem, please?

Abhiram824, 2024-11-25:
Hi, I tried running this code locally and was unable to reproduce the error. Could you provide more information about your installation of robosuite? Also please let me know if the following script works for you

```import numpy as np
import robosuite as suite
from robosuite.environments.manipulation.lift import Lift # replace with environment of choice

# pass in same args as before except for env_name
env = Lift(
    robots="Panda", 
    has_renderer=True,
    has_offscreen_renderer=False,
    use_camera_obs=False,
)

env.reset()
for i in range(1000):
    action = np.random.randn(*env.action_spec[0].shape) * 0.1
    obs, reward, done, info = env.step(action) # take action in the environment
    env.render() # render on display```

snasiriany, 2025-01-02:
Closing due to inactivity.

closed 2025-01-02 by snasiriany