ARISE-Initiative/robosuite #578: Finger joints not moving for GR1 with inspire hands

chaitanya1chawla, 2024-11-22:
### System Info

```Shell
Robosuite version: 1.5.0
```

### Information

    env = suite.make(
        env_name="Lift",
        robots="GR1ArmsOnly",
        gripper_types=["InspireRightHand", "InspireLeftHand"],
        env_configuration="bimanual",
        render_camera="agentview",
        controller_configs=load_controller_config(default_controller="OSC_POSE"), ##  args.controller_config
        has_renderer=True,
        has_offscreen_renderer=False,
        control_freq=20,
        ignore_done=True,
        use_camera_obs=False,
        reward_shaping=True,
        hard_reset=False,
    )
    env = VisualizationWrapper(env, indicator_configs=None)

    while True:
             obs=env.reset()
             env.render()
             flag=True
                   while flag=True:
                           action[:] += 0.001
                           obs, reward, done, info = env.step(action)
                           env.render()

### Reproduction

Not able to send finger joint commands in GR1ArmsOnly environment with inspire Hands.
Tried sending commands through **env.step(action)** as well as setting directly, via env.sim.data.ctrl[i] with a torque control.
Am I missing something?

xieleo5, 2024-11-26:
A more concrete example:
Just change the line to `load_composite_controller_config(robot="GR1")`, you should be able to see finger movements.

xieleo5, 2024-11-26:
Also, for dexterous hands. You need to make sure your controller config sets `use_action_scaling=false` like this:
```
"gripper": {
    "type": "GRIP",
    "use_action_scaling": false
}
```