ARISE-Initiative/robosuite #579: Support for bimanual robots with dexterous hands

chaitanya1chawla, 2024-11-23:
### System Info

```Shell
Robosuite version: 1.5.0
```

### Information

When spawning GR1 with inspire hands, the hands don't get correctly matched to each of the arms.

Upon some digging, `manipulation_env.py` makes a `gripper_types` array of length `num_robots`, whereas it should make an array of length `num_arms`, and subsequently each hand gets assigned to the respective arm.

The following code segment spawns a GR1 and assigns both hands as InspireRightHand.

```
env = suite.make(
    env_name="Lift",
    robots="GR1ArmsOnly",
    gripper_types=["InspireRightHand", "InspireLeftHand"],
    env_configuration="bimanual",
    render_camera="agentview",
    controller_configs=load_controller_config(default_controller="OSC_POSE"), ##  args.controller_config
    has_renderer=True,
    has_offscreen_renderer=False,
    control_freq=20,
    ignore_done=True,
    use_camera_obs=False,
    reward_shaping=True,
    hard_reset=False,
)
env = VisualizationWrapper(env, indicator_configs=None)

while True:
         obs=env.reset()
         env.render()
         flag=True
               while flag=True:
                       action[:] += 0.001
                       obs, reward, done, info = env.step(action)
                       env.render()
```

### Reproduction

Run above code segment

kevin-thankyou-lin, 2024-11-24:
@xieleo5 do you mind pointing @chaitanya1chawla to scripts for the GR1 that allows finger movement?

xieleo5, 2024-11-26:
Hi, @chaitanya1chawla. Can you try not to pass the gripper_types in your args? All the robots will have their default grippers; for the GR1 robot, it's defined [here](https://github.com/ARISE-Initiative/robosuite/blob/a651720a30c12599ec4f86a60d6fa277945f76ff/robosuite/models/robots/manipulators/gr1_robot.py#L25-L33). This can be a temporary solution for this issue. The argument may not work properly for humanoid robots, and we will investigate this.

chaitanya1chawla, 2024-11-26:
Thanks @xieleo5!
Our GR1 has inspire_hands attached to it, thus wanted the same setup in sim.
But thank you for the heads-up!

closed 2024-11-26 by chaitanya1chawla

xieleo5, 2024-11-26:
Yeah if you want to change to inspire hand, you can manually change in the default gripper in robot py for now.