ARISE-Initiative/robosuite #603: How to set camera intrinsics?

pumfish, 2025-01-05:
Hello，I’m trying add a new camera into `robosuite`, is there any way to set the camera intrinsics?
By the way, when I set the camera pos, I found that the documentation does not specify the length units used in robosuite. I would like to ask if robosuite has a specific length unit set? Or what is the scale ratio between the robots in robosuite and the robots in the real world?
robosuite version is `1.5.0`

pumfish, 2025-01-07:
I found and tried https://github.com/ARISE-Initiative/robosuite/blob/82accbb51fb05bd79c427edadf085a17be7b27aa/robosuite/utils/mjmod.py#L682 
But found it to be useless.

Abhiram824, 2025-01-07:
Hello, you can add a custom camera to your simulation by updating the simulation xml and adding a camera element. You can refer to the code here https://github.com/robocasa/robocasa/blob/main/robocasa/environments/kitchen/kitchen.py#L1019.

You can find more information about camera elements here: https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-camera.

Regarding unit conversion, I believe 1 unit in robosuite corresponds to 1 meter in the real world.

pumfish, 2025-01-09:
Thank you very much for your reply; my question has basically been resolved.

closed 2025-01-09 by pumfish