ARISE-Initiative/robosuite #615: Wired action control when using OSC_POSE controller and control_delta = False by version v1.4.1

yl-wang996, 2025-01-08:
### System Info

```Shell
OS: Ubuntu 20.04 
Robosuite Version: 1.4.1
```

### Information

## My problem
The following script should let the robot keep the pose as the first frame. But it seems to turn the TCP 45 degrees around the z-axis. Do you have any clue?
```
from robosuite.controllers import load_controller_config
from robosuite.utils.input_utils import *
from robosuite.utils.transform_utils import *

if __name__ == "__main__":

    options = {}

    options["env_name"] = "ToolHang"
    options["robots"]=["Panda"]
    options["controller_configs"] = load_controller_config(default_controller="OSC_POSE")

    # I set control_delta to False
    options["controller_configs"]["control_delta"] = False

    env = suite.make(
        **options,
        has_renderer=True,
        has_offscreen_renderer=False,
        ignore_done=True,
        use_camera_obs=False,
        control_freq=20,
    )
    obs = env.reset()
    env.viewer.set_camera(camera_id=0)
    low, high = env.action_spec

    # use the tcp at first frame as the action
    agent_pos = obs['robot0_eef_pos']
    agent_quat = obs['robot0_eef_quat']
    agent_rmat = quat2mat(agent_quat)
    agent_rot = quat2axisangle(mat2quat(agent_rmat))
    action = np.concatenate([agent_pos, agent_rot, [0]])
    print(action)

    for i in range(10000):
        obs, reward, done, _ = env.step(action)
        env.render()
```

### The robot pose in the first frame:
[image]
### The robot pose after the action executed:
[image]

### Reproduction

run the code:
'''
```
from robosuite.controllers import load_controller_config
from robosuite.utils.input_utils import *
from robosuite.utils.transform_utils import *

if __name__ == "__main__":

    options = {}

    options["env_name"] = "ToolHang"
    options["robots"]=["Panda"]
    options["controller_configs"] = load_controller_config(default_controller="OSC_POSE")

    # I set control_delta to False
    options["controller_configs"]["control_delta"] = False

    env = suite.make(
        **options,
        has_renderer=True,
        has_offscreen_renderer=False,
        ignore_done=True,
        use_camera_obs=False,
        control_freq=20,
    )
    obs = env.reset()
    env.viewer.set_camera(camera_id=0)
    low, high = env.action_spec

    # use the tcp at first frame as the action
    agent_pos = obs['robot0_eef_pos']
    agent_quat = obs['robot0_eef_quat']
    agent_rmat = quat2mat(agent_quat)
    agent_rot = quat2axisangle(mat2quat(agent_rmat))
    action = np.concatenate([agent_pos, agent_rot, [0]])
    print(action)

    for i in range(10000):
        obs, reward, done, _ = env.step(action)
        env.render()
```
'''

### Expected behavior

When executing an action the same as the pose in the first frame(using OCS_POSE and control_deta=false), the robot TCP should not move.

kevin-thankyou-lin, 2025-01-09:
Right, this is a bug that we semi-fixed in 1.5.0. To get the expected behavior, you'll want to set the action to `eef_quat_site` e.g. [here](https://github.com/ARISE-Initiative/robosuite/blob/2f66486e9cfff08ebeb8bfa1452accb9a8f82334/robosuite/robots/robot.py#L433) in `robosuite==1.5.0`.

yl-wang996, 2025-01-13:
thank you, :+1:   that helps a lot.     Refer to [here](https://github.com/ARISE-Initiative/robosuite/blob/2f66486e9cfff08ebeb8bfa1452accb9a8f82334/robosuite/robots/robot.py#L433) , it can be fixed by using the site pose rather original data:

```
    eff_site_name = "gripper0_grip_site"
    obs['robot0_eef_pos'] = env.sim.data.site_xpos[env.sim.model.site_name2id(eff_site_name)]
    obs['robot0_eef_quat'] = mat2quat(env.sim.data.site_xmat[env.sim.model.site_name2id(eff_site_name)].reshape(3, 3))
```

of course the `eff_site_name` could vary in different environments, I didn't try.

closed 2025-01-13 by yl-wang996

kevin-thankyou-lin, 2025-01-13:
Great to hear!