ARISE-Initiative/robosuite #616: ImportError

FuNz-0, 2025-01-09:
ImportError: Cannot initialize a EGL device display. This likely means that your EGL driver does not support the PLATFORM_DEVICE extension, which is required for creating a headless rendering context.

kevin-thankyou-lin, 2025-01-09:
Have you tried the solutions in #369 ?

Also, it'd be useful to have more context on your machine version etc

FuNz-0, 2025-01-13:
The reason is that my gpu does not support egl. Setting MUJOCO_GL=glx resolves this error. Would this affect the runing result?

kevin-thankyou-lin, 2025-01-13:
I see; glx should work for onscreen rendering, though mightn't work for offscreen rendering. I think osmesa should works for both on and off, though it's not GPU based so it's slower than using egl.

kevin-thankyou-lin, 2025-01-31:
Closing for now, let us know if you've more questions

closed 2025-01-31 by kevin-thankyou-lin