ARISE-Initiative/robosuite #621: Adding Mujoco-Menagerie XArm7

EdwardoSunny, 2025-01-14:
## System Info
```
Robosuite Version: v1.5.0
OS: Ubuntu 22.04
Mujoco: 3.2.6
Python: 3.10.12
```

## Problem Information
Similar to #605, I am trying to add a Xarm7 from [mujoco-menagerie](https://github.com/google-deepmind/mujoco_menagerie) as a custom imported arm. I largely followed the directions from the [importing model documentation](https://docs.google.com/document/d/1bSUKkpjmbKqWyV5Oc7_4VL4FGKAQZx8aWm_nvlmTVmE/edit?tab=t.0), with slightly modified steps for handling the XML as it was obtained from the mujoco-menagerie repo instead. 

The model is successfully loaded in robosuite and renders fine. The xml's are in the correct folders and renders correctly as well using `browse_mjcf_model.py`. I am able to attach camera observations as well. However, the arm's control seems to be wrong, as when I run the [tune_joints.py](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/scripts/tune_joints.py) script or any other script that asks the arm to hold its initial position, the arm just raises straight upwards and begins stuttering/shaking. 

Here's a visual of the problem (result directly after running `python tune_joints.py --robots XArm7`):
[image]

I also printed the joint positions in the loop of the tune_joints by adding this to the bottom of the `tune_joints.py`:
```
def get_joint_positions(joint_pos_cos, joint_pos_sin):
        return np.arctan2(joint_pos_sin, joint_pos_cos)  

    # just spin to let user interact with window
    while True:
        action = np.zeros(env.action_dim)
        obs, reward, done, _ = env.step(action)
        print(get_joint_positions(obs["robot0_joint_pos_cos"], obs["robot0_joint_pos_sin"]))
        env.render()
```

and the output seems to be raising joint positions over time:
```
[robosuite WARNING] Could not import robosuite_models. Some robots may not be available. If you want to use these robots, please install robosuite_models from source (https://github.com/ARISE-Initiative/robosuite_models) or through pip install. (__init__.py:30)

Welcome to the joint tuning script! You will be able to tune the robot
arm joints in the specified environment by using your keyboard. The
controls are printed below:

Keys      	Command
1-N       	Active Joint being tuned (N=number of joints for the active arm)
t         	Toggle between robot arms in the environment
r         	Reset active arm joints to all 0s
up/down   	incr/decrement the active joint angle

[robosuite INFO] Loading controller configuration from: /home/edward/Documents/Code/URIL/URIL-robosuite/robosuite/controllers/config/default/composite/basic.json (composite_controller_factory.py:121)
[robosuite WARNING] The config has defined for the controller "left", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for left from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "torso", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for torso from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "head", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for head from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "base", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for base from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "legs", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for legs from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "left", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for left from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "torso", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for torso from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "head", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for head from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "base", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for base from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "legs", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for legs from self.part_controller_config. (robot.py:151)
[0.000 -0.003 0.000 1.235 -0.000 1.233 -0.000]
[-0.000 0.000 -0.000 1.259 -0.000 1.256 -0.000]
[-0.000 0.004 -0.000 1.284 -0.000 1.278 -0.000]
[-0.000 0.006 -0.000 1.308 -0.000 1.300 -0.000]
[-0.000 0.009 -0.000 1.331 -0.000 1.321 -0.000]
[-0.000 0.012 -0.000 1.354 -0.000 1.341 -0.000]
[-0.000 0.016 -0.000 1.379 -0.000 1.364 -0.000]
[-0.000 0.019 0.000 1.403 -0.000 1.385 -0.000]
[-0.000 0.022 0.000 1.428 -0.000 1.406 -0.000]
[-0.000 0.027 0.000 1.454 -0.000 1.429 -0.000]
[-0.000 0.030 0.000 1.478 -0.000 1.449 -0.000]
[-0.000 0.034 0.000 1.502 -0.000 1.470 -0.000]
[-0.000 0.038 0.000 1.528 -0.000 1.492 -0.000]
[-0.000 0.041 0.000 1.550 -0.000 1.511 0.000]
[-0.000 0.045 0.000 1.575 -0.000 1.532 -0.000]
[-0.000 0.048 0.000 1.598 -0.000 1.552 -0.000]
[... 69 lines cut ...]
[robosuite WARNING] The config has defined for the controller "base", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for base from self.part_controller_config. (robot.py:151)
[robosuite WARNING] The config has defined for the controller "legs", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for legs from self.part_controller_config. (robot.py:151)
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
[-0.000 0.196 0.000 -2.618 0.000 2.942 0.785]
```

Similarly, if I try to run the teleportation script, whenever I give an input, there seems to be a "force" pulling the arm in a different random direction and the arm continuously stutters and tweaks.

## Reproduce Problem
My fork of the robosuite with my exact set up and xml files are at [this repo](https://github.com/EdwardoSunny/URIL-robosuite). Just set up the same way as regular robosuite and run the corresponding scripts to reproduce.

## Expected Behavior
Arm should stay still and hold position when running `tune_joints.py` without any additional input and should correctly move according to user input during teleportation.

EdwardoSunny, 2025-01-29:
I have resolved this issue. I will open a pull request soon for adding XArm7 :)

closed 2025-01-29 by EdwardoSunny

kevin-thankyou-lin, 2025-01-31:
Amazing, thank you!

shockstove, 2025-03-20:
hey, I also encountered this issue "there seems to be a "force" pulling the arm in a different random direction and the arm continuously stutters and tweaks." while importing my custom robot. 
I noticed your PR, but I would like to know if you could directly describe what changes you made to avoid the 'force'. 
Thank you very much!

shockstove, 2025-03-21:
My problem has been solved, and for some people who have encountered the same situation, investigating the following situations may be helpful:
Please note that the dual arm robots should be sorted in the order of RIGHT、LEFT in each file, even in the order of joints in the XML file. Otherwise, when tuning joints, they will automatically revert back to the init_pose specified in robot.exe after movement

shockstove, 2025-03-21:
> hey, I also encountered this issue "there seems to be a "force" pulling the arm in a different random direction and the arm continuously stutters and tweaks." while importing my custom robot. I [...]

My problem has been solved, and for some people who have encountered the same situation, investigating the following situations may be helpful:
Please note that the dual arm robots should be sorted in the order of RIGHT、LEFT in each file, even in the order of joints in the XML file. Otherwise, when tuning joints, they will automatically revert back to the init_pose specified in robot.exe after movement

EdwardoSunny, 2025-03-21:
> My problem has been solved, and for some people who have encountered the same situation, investigating the following situations may be helpful: Please note that the dual arm robots should be sorted [...]

Hi @shockstove , I apologize for the delay. But my solution was mainly looking into the XML file checking each joint parameter and ensuring that the dampening, force parameters, and the actuator control limit ranges are the right values corresponding to the ones from [mujoco-menagerie](https://github.com/google-deepmind/mujoco_menagerie). Since you have to refactor the original XML, its easy to leave certain things out on accident. For me,