ARISE-Initiative/robosuite #636: Rendering color wrong

menzz333, 2025-02-15:
[image]

Hello, I am encountering an issue with **rendering color errors**. My robosuite version is **1.2.**

[image]

I have meticulously examined the code in **env_robosuite.py**. The video rendered here has obtained the frame. I proceeded to draw this frame and discovered that the rendered color is still incorrect. I understand that this suggests the frame I received from robosuite might be problematic. Could you please advise on how to resolve this issue? Thank you.

abhihjoshi, 2025-02-20:
It seems the collision geoms are being rendered with the visual geoms. Can you try turning off the collision geoms during rendering?

menzz333, 2025-02-22:
> It seems the collision geoms are being rendered with the visual geoms. Can you try turning off the collision geoms during rendering?

Thank you for your help :). I tried to turn off the collision geoms during rendering, but then I was stuck by strange bugs. I only have gpu0 but it warned me that I do not have gpu. I install the package from [https://github.com/StanfordVL/egl_probe](url)

[image]

abhihjoshi, 2025-02-22:
What is the command you are using here? Are you manually setting which GPU to use in the command?

menzz333, 2025-02-22:
> What is the command you are using here? Are you manually setting which GPU to use in the command?

Hello, I did not manually set gpu to use before I met this bug. Then I tried to forcely use gpu0 by `export CUDA_VISIBLE_DEVICES=0`, but nothing changed.

For the command, I just used training script to train my model. In detail, I set _render_offscreen_ to _True_ to turn off the collision geoms. 

[image]

[image]