ARISE-Initiative/robosuite #637: Human Demonstrations link in Doc missing

KSB21ST, 2025-02-16:
Hello,
When I look at this [page](https://robosuite.ai/docs/algorithms/demonstrations.html), 
it says the .h5 demos should be in models/assets/demonstrations but in the robosuite repo, there is no folder named like this. 
Also, when I click [datasets collected with robosuite](https://arise-initiative.github.io/robomimic-web/docs/introduction/datasets.html#robosuite-hdf5-datasets) indicated in the document, it shows 404 page, which is strange..!
Does anyone know how to download these human demonstrations?

abhihjoshi, 2025-02-19:
Hello! I believe the demonstrations were removed since they were taking up some space in the repository (we should actually update this in the documentation as well). Instead, I think you can use the Robomimic dataset and convert them to Robosuite compatible format. More information on this can be found here: https://robomimic.github.io/docs/datasets/robomimic_v0.1.html

kevin-thankyou-lin, 2025-03-24:
Apologies for the delay: for datasets converted into robomimic format, use robomimic scripts e.g. `dataset_states_to_obs.py`. For datasets collected directly in robosuite (and not converted to robomimic), use robosuite scripts.

| what is the difference between robomimic and robosuite

Great question: robomimic builds on top of robosuite and has (reinforcement) learning algorithms/policy training/architecture implemented. See this image from robomimic's website

[image].

Robosuite contains the base environments, teleoperation devices, robot controllers, interface with the underlying mujoco simulator, etc.

Michaelszeng, 2026-03-05:
Hi, I also have a question about this. The [robomimic dataset download script](https://github.com/ARISE-Initiative/robomimic/blob/master/robomimic/scripts/download_datasets.py) or similarly, the [huggingface direct dataset download link](https://huggingface.co/datasets/robomimic/robomimic_datasets/tree/main/v1.5) appear to only apply to these tasks: `["lift", "can", "square", "transport", "tool_hang", "lift_real", "can_real", "tool_hang_real"]`.

Robosuite appears to offer other tasks (i.e. [table wiping](https://robosuite.ai/docs/modules/environments.html#table-wiping)). Are datasets for these non-robomimic tasks no longer available?