ARISE-Initiative/robosuite #640: IK_POSE input action

ByeongGyu-Kim, 2025-02-20:
`options = {
    "env_name": "Lift",  
    "robots": "Panda", 
}

controller_name = "IK_POSE"

arm_controller_config = suite.load_part_controller_config(default_controller=controller_name)
options["controller_configs"] = refactor_composite_controller_config(
    arm_controller_config, options["robots"], ["right"]
)

env = suite.make(
    **options,
    has_renderer=True,
    has_offscreen_renderer=False,
    ignore_done=True,
    use_camera_obs=False,
    control_freq=20,
)
obs = env.reset()

for i in range(100):
    start = time.time()
    obs, reward, done, info = env.step(zero_action)
    print("-------------------------------")
    eef_pos = obs["robot0_eef_pos"]
    eef_quat = obs["robot0_eef_quat"]
    eef_rpy = R.from_quat(eef_quat).as_euler('xyz')
    env.render()

`

How can I create a code that keeps the robot stationary at its initial position when using the IK_POSE controller? Additionally, how can the gripper be utilized in IK_POSE?

kevin-thankyou-lin, 2025-02-20:
You'll want to use "absolute"  in the controller configs; gripper action would be the last dim in `action`.

super-dashuaibi, 2025-02-23:
there is a  bug ，you can try to modify the function's return( compute_joint_positions )in ik.py, the modifytion is follows: 
lines : 300 and  301 
#dq = q_des  # hack for now
                return q_des

super-dashuaibi, 2025-02-23:
emmm,the robot can't stay still because the IK controller calculates the absolute position of the joint Angles, and then the JointPositionController receive it as a relative position, so you can try to set it just follows:
robosuite/controllers/parts/arm/ik.py lines 107  
adding the para:  input_type = "absolute",   
then it will stay

super-dashuaibi, 2025-02-23:
emmm,the robot can't stay still because the IK controller calculates the absolute position of the joint Angles, and then the JointPositionController receive it as a relative position, so you can try to set it just follows:
robosuite/controllers/parts/arm/ik.py lines 107
# Run sueprclass inits
        super().__init__(
            sim=sim,
            ref_name=ref_name,
            joint_indexes=joint_indexes,
            actuator_range=actuator_range,
            input_type = "absolute",  # this is the key 
            input_max=np.pi,
            input_min=-np.pi,
            output_max=np.pi,
            output_min=-np.pi,
            kp=kp,
            kv=kv,
            policy_freq=policy_freq,
            velocity_limits=[-1, 1],
            **kwargs,
        )
then it will stay

ByeongGyu-Kim, 2025-02-24:
> emmm,the robot can't stay still because the IK controller calculates the absolute position of the joint Angles, and then the JointPositionController receive it as a relative position, so you can try [...]

I have an additional question regarding IK_POSE and its action behavior. According to the documentation, it states:
**"When using any position-based control (OSC, IK, or JOINT_POSITION controllers), input actions are, by default, interpreted as delta values from the current state."**
This means that the following code should work by keeping the robot in its current position. However, when I run the code, the robot does not stay still. Why is this happening?

`options = {
    "env_name": "Lift", 
    "robots": "Panda",
}

controller_name = "IK_POSE"
arm_controller_config = suite.load_part_controller_config(default_controller=controller_name)
options["controller_configs"] = refactor_composite_controller_config(
    arm_controller_config, options["robots"], ["right"]
)
env = suite.make(
    **options,
    has_renderer=True,
    has_offscreen_renderer=False,
    ignore_done=True,
    use_camera_obs=False,
    control_freq=20,
)
obs = env.reset()
env.viewer.set_camera(camera_id=0)
zero_action = np.zeros(env.action_spec[0].shape)

print("action :", action)

for i in range(10000):
    obs, reward, done, _ = env.step(zero_action)
    env.render()
env.close()
`

kevin-thankyou-lin, 2025-02-26:
Good catch - I think that IK doesn't work correctly; however, you can try WHOLE_BODY_IK (absolute by default), or directly use this json (absolute by default):

`load_composite_controller_config(controller=<path/to/json/below>)`

```
{
    "type": "WHOLE_BODY_IK",
    "composite_controller_specific_configs": {
        "ref_name": ["gripper0_right_grip_site"],
        "interpolation": null,
        "actuation_part_names": ["right"],
        "max_dq": 4,
        "nullspace_joint_weights": {
            "robot0_joint1": 1,
            "robot0_joint2": 1,
            "robot0_joint3": 1,
            "robot0_joint4": 1,
            "robot0_joint5": 1,
            "robot0_joint6": 1,
            "robot0_joint7": 1
        },
        "ik_pseudo_inverse_damping": 5e-2,
        "ik_integration_dt": 1e-1,
        "ik_max_dq": 4.0,
        "ik_max_dq_torso": 0.2,
        "ik_input_type": "absolute",
        "ik_input_rotation_repr": "axis_angle",
        "verbose": false
    },
    "body_parts": {
        "arms": {
            "right": {
                "type" : "JOINT_POSITION",
                "input_max": 1,
                "input_min": -1,
                "input_type": "absolute",
                "output_max": 0.5,
                "output_min": -0.5,
                "kp": 500,
                "kd": 20,
                "velocity_limits": [-1,1],
                "kp_limits": [0, 300],
                "interpolation": null,
                "ramp_ratio": 0.2,
                "gripper": {
                    "type": "GRIP"
                }
            }
        }
    }
}

```

For WHOLE_BODY_IK, can you test setting `ik_input_type=delta` and `action=np.zeros(...)`

super-dashuaibi, 2025-02-27:
> > emmm,the robot can't stay still because the IK controller calculates the absolute position of the joint Angles, and then the JointPositionController receive it as a relative position, so you can try to set it just follows: robosuite/controllers/parts/arm/ik.py lines 107emmm，机器人不能保持静止，因为IK控制器计算关节角度的绝对位置，然后JointPositionController将其作为相对位置接收，因此您可以尝试将其设置如下：robosuite/controllers/parts/arm/ik.py行107
> > # Run superclass inits运行超类初始化
> > ```
> >     super().__init__(
> >         sim=sim,
> >         ref_name=ref_name,
> >         joint_indexes=joint_indexes,
> >         actuator_range=actuator_range,
> >         input_type = "absolute",  # this is the key 
> >         input_max=np.pi,
> >         input_min=-np.pi,
> >         output_max=np.pi,
> >         output_min=-np.pi,
> >         kp=kp,
> >         kv=kv,
> >         policy_freq=policy_freq,
> >         velocity_limits=[-1, 1],
> >         **kwargs,
> >     )
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > then it will stay那它就会留下来
> 
> I have an additional question regarding IK_POSE and its action behavior. According to the documentation, it states: **"When using any position-based control (OSC, IK, or JOINT_POSITION controllers), input actions are, by default, interpreted as delta values from the current state."** This means that the following code should work by keeping the robot in its current position. However, when I run the code, the robot does not stay still. Why is this happening?我还有一个关于IK_POSE及其动作行为的问题。根据文件，它指出：当使用任何基于位置的控制（OSC、IK或JOINT_POSITION控制器）时，默认情况下，输入动作被解释为当前状态的增量值。"这意味着下面的代码应该通过将机器人保持在其当前位置来工作。然而，当我运行代码时，机器人不会静止不动。为什么会发生这种情况？
> 
> ```
> options = {
>     "env_name": "Lift", 
>     "robots": "Panda",
> }
> 
> controller_name = "IK_POSE"
> arm_controller_config = suite.load_part_controller_config(default_controller=controller_name)
> options["controller_configs"] = refactor_composite_controller_config(
>     arm_controller_config, options["robots"], ["right"]
> )
> env = suite.make(
>     **options,
>     has_renderer=True,
>     has_offscreen_renderer=False,
>     ignore_done=True,
>     use_camera_obs=False,
>     control_freq=20,
> )
> obs = env.reset()
> env.viewer.set_camera(camera_id=0)
> zero_action = np.zeros(env.action_spec[0].shape)
> 
> print("action :", action)
> 
> for i in range(10000):As mentioned in my previous reply
>     obs, reward, done, _ = env.step(zero_action)
>     env.render()
> env.close()
> ```

You can try the above contributor's solution, I haven't tried it but I think it's the solution to your problem.

As for the problem you mentioned, the robotic arm cannot be stopped when action is 0 ,this is because, in the existing code, the joint Angle calculated by the InverseKinematicsController in the ik.py after receving the action is absolute angles, for example,we can assume it is  [1.0,1.0,1.0,1.0],But when InverseKinematicsController pass it to the super class,JointPositionController , the JointPositionController treats it as an increment,So the next step will emerge joint Angle is [1.0, 1.0, 1.0, 1.0, 1.0, 1.0] + [1.0, 1.0, 1.0, 1.0, 1.0, 1.0] = [2.0, 2.0, 2.0, 2.0, 2.0, 2.0] so the mechanical arm  can't stop even though the action is all zeros.

A super simple solution is to modify the super class's init in the ik .py .As mentioned in my previous reply:
robosuite/controllers/parts/arm/ik.py行107:
    super().__init__(
        sim=sim,
        ref_name=ref_name,
        joint_indexes=joint_indexes,
        actuator_range=actuator_range,
        input_type = "absolute",  # this is the key 
        input_max=np.pi,
        input_min=-np.pi,
        output_max=np.pi,
        output_min=-np.pi,
        kp=kp,
        kv=kv,
        policy_freq=policy_freq,
        velocity_limits=[-1, 1],
        **kwargs,
    )

Here, we specify by “input_type = "absolute"” that the joint Angle received by the JointPositionController is an absolute value rather than an incremental value.
And “ input_max=np.pi,
        input_min=-np.pi,
        output_max=np.pi,
        output_min=-np.pi,”ensures that the absolute joint Angle is not scaled

ByeongGyu-Kim, 2025-02-28:
Thank you all! I finally solve the problem!!

closed 2025-02-28 by ByeongGyu-Kim

auee028, 2025-04-21:
Thank you for helps!

Well I tried as following, but I have 90 degrees turn around z axis:
```
# Path to config file
controller_fpath = "../custom_utils/whole_body_ik.json"

# Import the file as a dict
controller_config = load_composite_controller_config(controller=controller_fpath)

env = suite.make(
    env_name="Stack",
    robots="UR5e",
    controller_configs=controller_config,
    has_renderer=False,
    has_offscreen_renderer=True,
)

obs = env.reset()

robot = env.robots[0]

eef_pos = obs["robot0_eef_pos"]
eef_quat = obs["robot0_eef_quat"]
# eef_quat = np.array([eef_quat[1], eef_quat[2], eef_quat[3], eef_quat[0]])  # wxyz to xyzw
eef_rpy = R.from_quat(eef_quat).as_euler('xyz')
gripper_state = [1.0]

for i in range(50):
    action = np.concatenate([eef_pos, eef_rpy, gripper_state])
    obs, reward, done, info = env.step(action)
```

Below is my `../custom_utils/whole_body_ik.json`:
```
{
    "type": "WHOLE_BODY_IK",
    "composite_controller_specific_configs": {
        "ref_name": ["gripper0_right_grip_site"],
        "interpolation": null,
        "actuation_part_names": ["right"],
        "max_dq": 4,
        "nullspace_joint_weights": {
            "robot0_torso_waist_yaw": 100.0,
            "robot0_torso_waist_pitch": 100.0,
            "robot0_torso_waist_roll": 500.0,
            "robot0_l_shoulder_pitch": 4.0,
            "robot0_r_shoulder_pitch": 4.0,
            "robot0_l_shoulder_roll": 3.0,
            "robot0_r_shoulder_roll": 3.0,
            "robot0_l_shoulder_yaw": 2.0,
            "robot0_r_shoulder_yaw": 2.0
        },
        "ik_pseudo_inverse_damping": 5e-2,
        "ik_integration_dt": 1e-1,
        "ik_max_dq": 4.0,
        "ik_max_dq_torso": 0.2,
        "ik_input_type": "absolute",
        "ik_input_rotation_repr": "axis_angle",
        "verbose": false
    },
    "body_parts": {
        "arms": {
            "right": {
                "type" : "JOINT_POSITION",
                "input_max": 1,
                "input_min": -1,
                "input_type": "absolute",
                "output_max": 0.5,
                "output_min": -0.5,
                "kp": 200,
                "kd": 200,
                "kp": 1000,
                "velocity_limits": [-1,1],
                "kp_limits": [0, 1000],
                "interpolation": null,
                "ramp_ratio": 0.2,
                "gripper": {
                    "type": "GRIP"
                }
            }
        }
    }
}
```

How can I keep from turning 90 degree and make the robot to stay still..?