ARISE-Initiative/robosuite #646: verify openvla in robosuite

shouji000, 2025-03-04:
### System Info

```Shell
robosuite 1.15.1
Ubuntu22.04
Jetson Agx orin(64G)
conda environment(python 3.10)
I want to verify openvla in robosuite
```

### Information

I wirte a simple code

```
import robosuite as suite
from robosuite.controllers import load_part_controller_config
from transformers import AutoModelForVision2Seq, AutoProcessor
from PIL import Image
import torch

local_model_path = "/home/yljy/jetson-containers/data/models/huggingface/models--openvla--openvla-7b/snapshots/31f090d05236101ebfc381b61c674dd4746d4ce0"

processor = AutoProcessor.from_pretrained(local_model_path, trust_remote_code=True)
vla = AutoModelForVision2Seq.from_pretrained(
    local_model_path,
    attn_implementation="flash_attention_2",
    torch_dtype=torch.bfloat16,
    low_cpu_mem_usage=True,
    trust_remote_code=True
).to("cuda:0")

controller_config = load_part_controller_config(default_controller="IK_POSE")

robosuite_env = suite.make(
    "Lift",
    robots="Panda",
    has_renderer=True,
    has_offscreen_renderer=True,
    use_camera_obs=True,
    camera_names="frontview",
    camera_heights=640,
    camera_widths=480
)

obs = robosuite_env.reset()

prompt = "In: What action should the robot take to pick up the cube?\nOut:"

while(True):
    image = Image.fromarray(obs['frontview_image'])

    inputs = processor(prompt, image).to("cuda:0", dtype=torch.bfloat16)
    action = vla.predict_action(**inputs, unnorm_key="bridge_orig", do_sample=False)
    action[0:3] = action[0:3]*100 #because the sensitivity is 0.01
    action[6] = action[6]*2-1 #related to the gripper openvla is [0,1],robosuite is [-1,1]

    print(action)

    obs, reward, done, info = robosuite_env.step(action)
    robosuite_env.render()
```

### Reproduction

I run the code to use the openvla to control the manipulator, but something goes wrong. By rights, the robotic arm should be moving down, but the action[2] that openvla outputs (i.e., the offset of the Z-axis) is always positive. Is it because the end coordinates in robosuite don't match the end coordinates in openvla?

[image]

ByeongGyu-Kim, 2025-03-16:
Did you solve this problem? I also struggling with this problem.

shouji000, 2025-03-24:
> it seems that openvla-7b was trained on the bridgedataV2, which is based on wx250(or vx300), there is a gap between franka panda and wx250. Camera and environment also influence openVLA. So, I don't think this is a problem caused by Robosuite. Maybe you can try SimplerEnv which is based on Sapien. Someone has pushed his code for verification of openVLA on GIthub

Yeah！I have use the mimicgen data to finetune. The task is Stack. But I don't know how to verify it

shouji000, 2025-03-24:
> Did you solve this problem? I also struggling with this problem.

I have not sloved the problem
I try to finetune it. Do you solve it?

snasiriany, 2025-03-24:
Hi, as @goodrain553 suggested, open-vla was not trained on Robosuite datasets, so it may not work out of the box on RoboSuite tasks. I can share the semantics of the robosuite action space for the Panda robot:

Assuming a 0-indexed action array:
dim 0: forward / back
dim 1: right / left
dim 2: up / down

dim 3: roll
dim 4: pitch
dim 5: yaw

dim 6: gripper

You may need to fine-tune OpenVLA on RoboSuite data, but if that does not work, that may be more with the OpenVLA model side rather than with the RoboSuite side. Hope this helps!

closed 2025-03-24 by snasiriany

lorepieri8, 2025-05-22:
@shouji000 What did you ended up doing?