ARISE-Initiative/robosuite #652: Demo Playback for robocasa Broken

shaoyifei96, 2025-03-10:
### System Info

```Shell
Ubuntu 22.04
```

### Information

At commit dd895e4903a40eb7f8d2d4db848d57dc26197df4 , robocasa demo playback were fine. With the recent updates, things don't work anymore.

```
$ python robocasa/robocasa/demos/demo_tasks.py --task OpenSingleDoor 
[robosuite WARNING] No private macro file found! (macros.py:57)
[robosuite WARNING] It is recommended to use a private macro file (macros.py:58)
[robosuite WARNING] To setup, run: python /home/yifei/Documents/task_planning/robosuite/robosuite/scripts/setup_macros.py (macros.py:59)
[robosuite WARNING] Could not import robosuite_models. Some robots may not be available. If you want to use these robots, please install robosuite_models from source (https://github.com/ARISE-Initiative/robosuite_models) or through pip install. (__init__.py:30)
WARNING: mimicgen environments not imported since mimicgen is not installed!
Initializing environment for OpenSingleDoor...
[robosuite INFO] Loading controller configuration from: /home/yifei/Documents/task_planning/robosuite/robosuite/controllers/config/robots/default_pandaomron.json (composite_controller_factory.py:121)

Playing back episode: demo_1
Instruction: open the cabinet door
Spawning environment...
Traceback (most recent call last):
  File "/home/yifei/Documents/task_planning/robocasa/robocasa/demos/demo_tasks.py", line 150, in <module>
    playback_dataset(parser)
  File "/home/yifei/Documents/task_planning/robocasa/robocasa/scripts/playback_dataset.py", line 397, in playback_dataset
    playback_trajectory_with_env(
  File "/home/yifei/Documents/task_planning/robocasa/robocasa/scripts/playback_dataset.py", line 63, in playback_trajectory_with_env
    reset_to(env, initial_state)
  File "/home/yifei/Documents/task_planning/robocasa/robocasa/scripts/playback_dataset.py", line 251, in reset_to
    env.reset_from_xml_string(xml)
  File "/home/yifei/Documents/task_planning/robosuite/robosuite/environments/base.py", line 624, in reset_from_xml_string
    self.reset()
  File "/home/yifei/Documents/task_planning/robosuite/robosuite/environments/base.py", line 282, in reset
    self._reset_internal()
  File "/home/yifei/Documents/task_planning/robocasa/robocasa/environments/kitchen/single_stage/kitchen_doors.py", line 65, in _reset_internal
    super()._reset_internal()
  File "/home/yifei/Documents/task_planning/robocasa/robocasa/environments/kitchen/kitchen.py", line 883, in _reset_internal
    super()._reset_internal()
  File "/home/yifei/Documents/task_planning/robosuite/robosuite/environments/robot_env.py", line 519, in _reset_internal
    super()._reset_internal()
  File "/home/yifei/Documents/task_planning/robosuite/robosuite/environments/base.py", line 340, in _reset_internal
    self._setup_references()
  File "/home/yifei/Documents/task_planning/robocasa/robocasa/environments/kitchen/kitchen.py", line 1176, in _setup_references
    super()._setup_references()
  File "/home/yifei/Documents/task_planning/robosuite/robosuite/environments/robot_env.py", line 322, in _setup_references
    super()._setup_references()
  File "/home/yifei/Documents/task_planning/robosuite/robosuite/environments/base.py", line 226, in _setup_references
    self.model.generate_id_mappings(sim=self.sim)
  File "/home/yifei/Documents/task_planning/robosuite/robosuite/models/tasks/task.py", line 153, in generate_id_mappings
    body_id = sim.model.body_name2id(body_name)
  File "/home/yifei/Documents/task_planning/robosuite/robosuite/utils/binding_utils.py", line 367, in body_name2id
    raise ValueError('No "body" with name %s exists. Available "body" names = %s.' % (name, self.body_names))
ValueError: No "body" with name left_eef_target exists. Available "body" names = ('world', 'robot0_base', 'mobilebase0_base', 'mobilebase0_fixed_support', 'mobilebase0_support', 'robot0_link0', 'robot0_link1', 'robot0_link2', 'robot0_link3', 'robot0_link4', 'robot0_link5', 'robot0_link6', 'robot0_link7', 'robot0_right_hand', 'gripper0_right_right_gripper', 'gripper0_right_eef', 'gripper0_right_leftfinger', 'gripper0_right_finger_joint1_tip', 'gripper0_right_rightfinger', 'gripper0_right_finger_joint2_tip', 'mobilebase0_wheeled_base', 'wall_room_main', 'wall_backing_room_main', 'wall_left_room_main', 'wall_left_backing_room_main', 'outlet_room_main', 'outlet_2_room_main', 'light_switch_room_main', 'light_switch_2_room_main', 'floor_room_main', 'floor_backing_room_main', 'counter_corner_main_main_group_main', 'counter_main_main_group_main', 'stovetop_main_group_main', 'stovetop_main_group_knob_rear_right', 'stovetop_main_group_knob_front_left', 'stovetop_main_group_knob_front_right', 'stovetop_main_group_knob_rear_left', 'counter_1_right_main_group_main', 'fridge_main_group_main', 'fridge_housing_main_group_main', 'stack_1_main_group_base_main', 'stack_1_main_group_1_main', 'stack_1_main_group_1_inner_box', 'stack_1_main_group_1_door_main', 'stack_1_main_group_1_door_trims', 'stack_1_main_group_1_door_handle_main', 'stack_1_main_group_2_main', 'stack_1_main_group_2_inner_box', 'stack_1_main_group_2_door_main', 'stack_1_main_group_2_door_trims', 'stack_1_main_group_2_door_handle_main', 'stack_1_main_group_3_main', 'stack_1_main_group_3_inner_box', 'stack_1_main_group_3_door_main', 'stack_1_main_group_3_door_trims', 'stack_1_main_group_3_door_handle_main', 'stack_1_main_group_4_main', 'stack_1_main_group_4_inner_box', 'stack_1_main_group_4_door_main', 'stack_1_main_group_4_door_trims', 'stack_1_main_group_4_door_handle_main', 'stack_2_main_group_base_main', 'stack_2_main_group_1_main', 'stack_2_main_group_1_hingeleftdoor', 'stack_2_main_group_1_left_door_main', 'stack_2_main_group_1_left_door_trims', 'stack_2_main_group_1_left_door_handle_main', 'stack_2_main_group_1_hingerightdoor', 'stack_2_main_group_1_right_door_main', 'stack_2_main_group_1_right_door_trims', 'stack_2_main_group_1_right_door_handle_main', 'stack_2_main_group_2_main', 'stack_2_main_group_2_door_main', 'stack_2_main_group_2_door_trims', 'stack_3_main_group_base_main', 'stack_3_main_group_1_main', 'stack_3_main_group_1_hingeleftdoor', 'stack_3_main_group_1_left_door_main', 'stack_3_main_group_1_left_door_trims', 'stack_3_main_group_1_left_door_handle_main', 'stack_3_main_group_1_hingerightdoor', 'stack_3_main_group_1_right_door_main', 'stack_3_main_group_1_right_door_trims', 'stack_3_main_group_1_right_door_handle_main', 'stack_3_main_group_2_main', 'stack_3_main_group_2_inner_box', 'stack_3_main_group_2_door_main', 'stack_3_main_group_2_door_trims', 'stack_3_main_group_2_door_handle_main', 'stack_3_main_group_3_main', 'stack_3_main_group_3_inner_box', 'stack_3_main_group_3_door_main', 'stack_3_main_group_3_door_trims', 'stack_3_main_group_3_door_handle_main', 'hood_main_group_main', 'cab_main_main_group_main', 'cab_main_main_group_hingeleftdoor', 'cab_main_main_group_left_door_main', 'cab_main_main_group_left_door_trims', 'cab_main_main_group_left_door_handle_main', 'cab_main_main_group_hingerightdoor', 'cab_main_main_group_right_door_main', 'cab_main_main_group_right_door_trims', 'cab_main_main_group_right_door_handle_main', 'shelves_main_group_main', 'shelves_main_group_hingecab', 'shelves_main_group_shelf_0_main', 'shelves_main_group_shelf_1_main', 'fridge_cab_main_group_main', 'fridge_cab_main_group_hingeleftdoor', 'fridge_cab_main_group_left_door_main', 'fridge_cab_main_group_left_door_trims', 'fridge_cab_main_group_left_door_handle_main', 'fridge_cab_main_group_hingerightdoor', 'fridge_cab_main_group_right_door_main', 'fridge_cab_main_group_right_door_trims', 'fridge_cab_main_group_right_door_handle_main', 'toaster_main_group_main', 'utensil_holder_main_group_main', 'cab_1_left_group_main', 'cab_1_left_group_hingedoor', 'cab_1_left_group_door_main', 'cab_1_left_group_door_trims', 'cab_1_left_group_door_handle_main', 'window_group_left_group_root', 'window_group_left_group_trim_left_0_main', 'window_group_left_group_trim_right_0_main', 'window_group_left_group_trim_top_0_main', 'window_group_left_group_trim_bottom_0_main', 'window_group_left_group_vert_trim_0_main', 'window_group_left_group_horiz_trim_0_main', 'window_group_left_group_trim_left_1_main', 'window_group_left_group_trim_right_1_main', 'window_group_left_group_trim_top_1_main', 'window_group_left_group_trim_bottom_1_main', 'window_group_left_group_vert_trim_1_main', 'window_group_left_group_horiz_trim_1_main', 'window_group_left_group_trim_left_2_main', 'window_group_left_group_trim_right_2_main', 'window_group_left_group_trim_top_2 [... cut ...] inner_box', 'stack_3_left_group_3_door_main', 'stack_3_left_group_3_door_trims', 'stack_3_left_group_3_door_handle_main', 'coffee_machine_left_group_main', 'paper_towel_left_group_main', 'knife_block_left_group_main', 'plant_left_group_main', 'stool_1_stool_group_main', 'stool_2_stool_group_main', 'stool_3_stool_group_main', 'door_obj_main', 'distr_counter_1_main', 'distr_counter_2_main', 'distr_counter_3_main').
```

### Reproduction

Install robocasa and robosuite per instruction of both repos

shaoyifei96, 2025-03-10:
offending commit seems to be 2f660c1008b2b786308a83f574428dae7b0aef62

kevin-thankyou-lin, 2025-03-13:
Good catch: could you please try this [hot-fix](https://github.com/ARISE-Initiative/robosuite/pull/653)?

kevin-thankyou-lin, 2025-03-13:
Oops, looks like I already made a PR [here](https://github.com/ARISE-Initiative/robosuite/pull/650) but hasn't been merged yet --- @Abhiram824 could you please review this PR? Thanks

kevin-thankyou-lin, 2025-03-24:
Resolved in #650

closed 2025-03-24 by kevin-thankyou-lin