ARISE-Initiative/robosuite #658: Run failed

Xotm, 2025-03-19:
### System Info

```Shell
Operating system: Ubuntu 24.04.2
robosuite Version: 1.5.1
```

### Information

Execute command: python -m robosuite.demos.demo_random_action

Get the result: GLFWError: (65542) b'EGL: Failed to get EGL display: Success'
  warnings.warn(message, GLFWError)
ERROR: could not create window

### Reproduction

1. python -m robosuite.demos.demo_random_action

abhihjoshi, 2025-03-19:
Could you try setting `MUJOCO_GL=egl` at the beginning of the command?
`MUJOCO_GL=egl python -m robosuite.demos.demo_random_action`

Xotm, 2025-03-20:
> 您可以尝试`MUJOCO_GL=egl`在命令开头进行设置吗？ `MUJOCO_GL=egl python -m robosuite.demos.demo_random_action`

I tried the method you suggested, but unfortunately I still get the same error. My device only has an integrated graphics card with 1GB of video memory, but I can successfully run the mujoco and OpenGl examples

kevin-thankyou-lin, 2025-03-21:
`GLFWError: (65542) b'EGL: Failed to get EGL display: Success'`

Sounds like your machine isn't connected to a display?