ARISE-Initiative/robosuite #659: Inconsistent Behavior in robosuite When Using Renderer

DeliYuxiang, 2025-03-19:
### System Info

```Shell
[project]
name = "robosuite"
version = "1.5.1"
description = ""
authors = [
    {name = "Your Name",email = "you@example.com"}
]
readme = "README.md"
requires-python = ">=3.12"
dependencies = [
    "pytest (>=8.3.5,<9.0.0)",
    "termcolor (>=2.5.0,<3.0.0)",
    "matplotlib (>=3.10.1,<4.0.0)",
    "tqdm (>=4.67.1,<5.0.0)",
    "mujoco (>=3.3.0,<4.0.0)",
    "opencv-python (>=4.11.0.86,<5.0.0.0)",
    "numpy (==2.1)",
    "numba (>=0.61.0,<0.62.0)",
    "scikit-learn (>=1.6.1,<2.0.0)",
    "h5py (>=3.13.0,<4.0.0)"
]

[build-system]
requires = ["poetry-core>=2.0.0,<3.0.0"]
build-backend = "poetry.core.masonry.api"
```

### Information

**Description:**
Running the `robosuite` environment with and without the renderer results in inconsistent behavior, as observed in multiple test cases. The test script attempts to verify consistency in rewards and observations across different environments and robot configurations. However, failures indicate discrepancies when the renderer is enabled (`has_offscreen_renderer=True`) versus when it is disabled (`has_offscreen_renderer=False`).

**Test Cases:**
1. **Reward Mismatch:**
   - For each environment-robot combination, the test executes a series of actions and compares the rewards returned by the two instances (one with the renderer, one without).
   - Assertion failure occurs when `reward != reward2`, implying that the same action leads to different rewards depending on the renderer setting.

2. **Observation Mismatch:**
   - The test ensures that observations remain consistent between the two instances.
   - Assertion failure occurs when key-wise observation values do not match between the two environments.
   - The test logs inconsistencies in specific observation keys, indicating deviations in environment states despite identical input actions.

**Actual Behavior:**
- Discrepancies in rewards and observations suggest that enabling the renderer modifies the environment's internal state or simulation dynamics.

### Reproduction

**Environment Information:**
- `robosuite` version: [commit:76842f918f16dab5582a128e06b262fc74a70700]
- Python version: [python 3.12]
- OS: [redhat]
- Any additional relevant details.

**Steps to Reproduce:**
1. Run the provided test script using `unittest`.
2. Observe assertion failures related to inconsistent rewards and observations.
3. Review logs for specific observation keys exhibiting discrepancies.

DeliYuxiang, 2025-03-19:
```python
import numpy as np
import unittest

import robosuite as suite
from pathlib import Path
import itertools
from robosuite.environments.manipulation.manipulation_env import ManipulationEnv

cwd = Path.cwd()

def get_envs(env_name: str, robot: str):
    env_config = {"env_name": env_name, "robots": robot}

    share_configs = {
        **env_config,
        "has_renderer": False,
        "use_object_obs": True,
        "reward_shaping": True,
        "use_camera_obs": False,
    }
    # Notify user of which test we are currently on
    print(f"Testing env: {env_name} with robot: {robot}...")
    env: ManipulationEnv = suite.make(
        **share_configs,
        has_offscreen_renderer=False,
    )
    render_env: ManipulationEnv = suite.make(
        **share_configs,
        has_offscreen_renderer=True,
    )

    return env, render_env

class TestAllCombinations(unittest.TestCase):

    def test_env_robot_combinations(self):
        envs = sorted(suite.ALL_ENVIRONMENTS)
        robots = [
            "Panda",
            "Sawyer",
            "Baxter",
            "GR1",
        ]

        # for env_name in envs:
        #     for robot_name in robots:
        for env_name, robot_name in itertools.product(envs, robots):
            with self.subTest(env=env_name, robot=robot_name):
                # Create config dict
                env, render_env = get_envs(env_name, robot_name)
                env.reset()
                render_env.reset()

                for i in range(env.horizon):
                    action = np.random.rand(*env.action_spec[0].shape)
                    action = env.action_spec[0] * action + env.action_spec[
                        1] * (1 - action)
                    # input(action)
                    obs, reward, *_, info = env.step(action)

                    obs1, reward2, *_, info2 = render_env.step(action)
                    assert reward == reward2, f"Inconsistant reward Step {i:04d} {reward:6e} {reward2:6e}"

            with self.subTest(env=env_name, robot=robot_name):
                # Create config dict
                env, render_env = get_envs(env_name, robot_name)
                obs = env.reset()
                obs1 = render_env.reset()
                join_keys = set(obs.keys()).intersection(obs1.keys())

                for i in range(env.horizon):
                    key_consistancy = True
                    keys = []
                    action = np.random.rand(*env.action_spec[0].shape)
                    action = env.action_spec[0] * action + env.action_spec[
                        1] * (1 - action)
                    # input(action)
                    obs, reward, *_, info = env.step(action)

                    obs1, reward2, *_, info2 = render_env.step(action)
                    for key in join_keys:
                        key_bool = np.all(obs[key] == obs1[key])
                        if not key_bool:
                            # print(
                            #     f"Inconsistant obs term {key}: {obs[key]} {obs1[key]}"
                            # )
                            keys.append(key)
                        key_consistancy = key_consistancy and key_bool
                    assert key_consistancy, f"Inconsistant obs term Step {i:4d} {keys}"

if __name__ == '__main__':
    unittest.main()

```

Abhiram824, 2025-03-24:
Apologies, I am unable to find the test script, could you point me to it? 

Also, what tasks are you doing this on? The reward/observation can be heavily influenced by the initialization of objects which are randomized every time an env is initalized/reset. This likely explains the inconsisten behavior

Abhiram824, 2025-04-21:
Closing this for now; feel free to open back up if you have any follow-ups.

closed 2025-04-21 by Abhiram824