ARISE-Initiative/robosuite #671: AssertionError: JOINT_POSITION controller is specified, but not imported or loaded

XieJunr, 2025-04-05:
### System Info

```Shell
Robosuite 1.5.0, Ubuntu20.04
```

### Information

config = {
    "type": "JOINT_POSITION",
    "input_max": 1,
    "input_min": -1,
    "output_max": 0.05,
    "output_min": -0.05,
    "kp": 50,
    "damping_ratio": 1,
    "impedance_mode": "fixed",
    "kp_limits": [0, 300],
    "damping_ratio_limits": [0, 10],
    "qpos_limits": "null",
    "interpolation": "null",
    "ramp_ratio": 0.2
  }

env = robosuite.make(
    "Door",
    controller_configs=config
)

Error:
assert type in REGISTERED_COMPOSITE_CONTROLLERS_DICT, f"{type} controller is specified, but not imported or loaded"
AssertionError: JOINT_POSITION controller is specified, but not imported or loaded

### Reproduction

1

### Expected behavior

I want to use the JOINT_POSITION controller, but it only supports composite_controller. What should I do

Abhiram824, 2025-04-06:
From robosuite v1.5 onward, all controllers need to be composite controllers, which assumes that a robot consists of multiple “body parts”, such as arms, torso, head, base, and legs, and that each body part has a “body part” controller (e.g., OSC_POSE, JOINT_POSITION). If you would like to control the robot arms using using a JOINT_POSITION controller, you can create a new config which specifies that. Might help to look at  the one of our premade [configs](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/config/default/composite/basic.json) for reference on creating composite controllers. More information on the controller structure can be found in our [documentation](https://robosuite.ai/docs/modules/controllers.html)

XieJunr, 2025-04-07:
> 从 robosuite v1.5 开始，所有控制器都需要是复合控制器，假设机器人由多个“身体部位”组成，例如手臂、躯干、头部、底座和腿部，并且每个身体部位都有一个“身体部位”控制器（例如，OSC_POSE、JOINT_POSITION）。如果您想使用 JOINT_POSITION 控制器来控制机器人手臂，您可以创建一个新的配置来指定该配置。查看我们的预制[配置](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/config/default/composite/basic.json)之一可能会有所帮助，以参考创建复合控制器。有关控制器结构的更多信息，请参阅我们的[文档](https://robosuite.ai/docs/modules/controllers.html)

closed 2025-04-07 by XieJunr