ARISE-Initiative/robosuite #685: WHOLE_BODY_IK input action

auee028, 2025-04-21:
Thank you for helps in #640!

Well I tried as following, but I have 90 degrees turn around z axis:
```
# Path to config file
controller_fpath = "../custom_utils/whole_body_ik.json"

# Import the file as a dict
controller_config = load_composite_controller_config(controller=controller_fpath)

env = suite.make(
    env_name="Stack",
    robots="UR5e",
    controller_configs=controller_config,
    has_renderer=False,
    has_offscreen_renderer=True,
)

obs = env.reset()

robot = env.robots[0]

eef_pos = obs["robot0_eef_pos"]
eef_quat = obs["robot0_eef_quat"]
# eef_quat = np.array([eef_quat[1], eef_quat[2], eef_quat[3], eef_quat[0]])  # wxyz to xyzw
eef_rpy = R.from_quat(eef_quat).as_euler('xyz')
gripper_state = [1.0]

for i in range(50):
    action = np.concatenate([eef_pos, eef_rpy, gripper_state])
    obs, reward, done, info = env.step(action)
```

Below is my `../custom_utils/whole_body_ik.json`:
```
{
    "type": "WHOLE_BODY_IK",
    "composite_controller_specific_configs": {
        "ref_name": ["gripper0_right_grip_site"],
        "interpolation": null,
        "actuation_part_names": ["right"],
        "max_dq": 4,
        "nullspace_joint_weights": {
            "robot0_torso_waist_yaw": 100.0,
            "robot0_torso_waist_pitch": 100.0,
            "robot0_torso_waist_roll": 500.0,
            "robot0_l_shoulder_pitch": 4.0,
            "robot0_r_shoulder_pitch": 4.0,
            "robot0_l_shoulder_roll": 3.0,
            "robot0_r_shoulder_roll": 3.0,
            "robot0_l_shoulder_yaw": 2.0,
            "robot0_r_shoulder_yaw": 2.0
        },
        "ik_pseudo_inverse_damping": 5e-2,
        "ik_integration_dt": 1e-1,
        "ik_max_dq": 4.0,
        "ik_max_dq_torso": 0.2,
        "ik_input_type": "absolute",
        "ik_input_rotation_repr": "axis_angle",
        "verbose": false
    },
    "body_parts": {
        "arms": {
            "right": {
                "type" : "JOINT_POSITION",
                "input_max": 1,
                "input_min": -1,
                "input_type": "absolute",
                "output_max": 0.5,
                "output_min": -0.5,
                "kp": 200,
                "kd": 200,
                "kp": 1000,
                "velocity_limits": [-1,1],
                "kp_limits": [0, 1000],
                "interpolation": null,
                "ramp_ratio": 0.2,
                "gripper": {
                    "type": "GRIP"
                }
            }
        }
    }
}
```

How can I keep from turning 90 degree and make the robot to stay still..?

kevin-thankyou-lin, 2025-04-21:
I think you can do `eef_quat = obs["robot0_eef_quat_site"]`. The 90 degree rotation is because the controller controls `gripper0_right_grip_site` whereas the observation's `eef_quat = obs["robot0_eef_quat"]` is a `body` (which I think has a 90 degree offset from the body).

auee028, 2025-04-21:
Thank you for your response! 

Additionally I found my error that I used Euler Angle with the scipy helper function, which caused the mismatch from what I expected. I fixed the code by switching to **Axis Angle** using `quat2axisangle()` of `robosuite.utils.transform_utils`, and it not the robot holds still without any translation or rotation when the target pose is the same with the current pose.
Here is the updated code:
```
# Path to config file
controller_fpath = "../custom_utils/whole_body_ik.json"

# Import the file as a dict
controller_config = load_composite_controller_config(controller=controller_fpath)

env = suite.make(
    env_name="Stack",
    robots="UR5e",
    controller_configs=controller_config,
    has_renderer=False,
    has_offscreen_renderer=True,
)

obs = env.reset()

robot = env.robots[0]

eef_pos = obs["robot0_eef_pos"]
eef_quat = obs["robot0_eef_quat_site"]
eef_rpy = robosuite.utils.transform_utils.quat2axisangle(eef_quat)   # key fix: conversion from quat to axis-angle
gripper_state = [1.0]

for i in range(50):
    action = np.concatenate([eef_pos, eef_rpy, gripper_state])
    obs, reward, done, info = env.step(action)
```

kevin-thankyou-lin, 2025-04-23:
Great! You're welcome!

closed 2025-04-23 by kevin-thankyou-lin