ARISE-Initiative/robosuite #686: Use the "Sphere" object as marker to display robot trajectory

vib2810, 2025-04-21:
Hi,
I have been trying to use the `Sphere` object to display things like robot trajectory, and I am not sure how I can set the position of the sphere visual object I create programmatically. As a proof of concept, I have been trying to overload the step function like this:

```
    def step(self, *args, **kwargs):
        noise = np.random.uniform(0, 0.3, size=3)
        new_pos = np.array([1.0, 0, 1.0]) + noise
        self.sphere.set('pos', new_pos.tolist())
        return super().step(*args, **kwargs)
```

But unfortunately, the sphere does not move. I am adding the sphere to the self.visual_objects list in the `_construct_visual_objects` function like this:
```
    def _construct_visual_objects(self):
       .... original code...
       self.sphere = BallObject(
            name="sphere",
            size=[0.04],
            rgba=[0, 0.5, 0.5, 1])
        self.visual_objects.append(self.sphere)
        self.sphere = self.sphere.get_obj()
        self.sphere.set('pos', '1.0 0 1.0')
```

Can you please tell me how can I update the code to allow me to **set the position of the sphere before the step function**

Abhiram824, 2025-04-21:
You could add a sphere site in the xml body of the end-effector of your robot. The site will move as the end-effector moves since its inside the end-effector's xml body (effectively tracking your trajectory). We actually do something similar for the Panda gripper for visualization. See [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/grippers/panda_gripper.xml#L21 )

Abhiram824, 2025-04-21:
Also you can refer to this issue which had a similar question: https://github.com/ARISE-Initiative/robosuite/issues/649

vib2810, 2025-04-21:
```
 Based on these two mechanisms of how MJCF models are created, we offer two main ways of creating your own object:
- Define an object in an MJCF XML file;
- Use procedural generation APIs to dynamically create an MJCF model.
```
Is it possible to do the same without modifying the XML and just adding the object using the dynamic API?

Abhiram824, 2025-04-21:
Sites are basically objects which are purely visual/dont participate in collisions. And yes you can dynamically add a site to the xml, see [here](https://github.com/robocasa/robocasa/blob/main/robocasa/environments/kitchen/kitchen.py#L847)

Abhiram824, 2025-05-31:
Closing for now. Feel free to open back up if you have any follow-up questions.

closed 2025-05-31 by Abhiram824