ARISE-Initiative/robosuite #687: New XArm7 Controller Issue

dayiethan, 2025-04-22:
I have previously used the OSC_POSE controller that follows waypoints to generate a trajectory for the Kinova3 arm. However, I want to generate these trajectories on the newly added XArm7, but it seems like there is no default controller available for the XArm7. Will this be added soon, or is there a workaround? Thanks!

Abhiram824, 2025-04-22:
You should be able to use the [basic controller](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/config/default/composite/basic.json) for the new robot

dayiethan, 2025-04-22:
That is the controller that I have been using for the Kinova arm, but when I try to use that with the XArm7, it produces erratic behavior. Are there other adaptations that need to be made to switch between the two?

Abhiram824, 2025-04-22:
Hi could you further detail how you got this erratic behavior/share any relevant scripts for reproduction? I ran `python robosuite/scripts/collect_human_demonstrations.py --environment Lift --robots XArm7 --device keyboard --controller BASIC` and  got pretty reasonable behavior.

closed 2025-06-06 by Abhiram824