ARISE-Initiative/robosuite #692: CUDA_VISIBLE_DEVICES does not work

zaizailu, 2025-05-04:
Hi! I am using robosuite for my rl work. I have 8 gpus. I used `` export CUDA_VISIBLE_DEVICES = 1`` , and when I use `` suite.make() ``, it just create someting about 500MB on my gpu2. What more strange is, when I use `` export CUDA_VISIBLE_DEVICES = 0``,  `` suite.make() ``creates someting about 500MB on my gpu3. I hope there is a way to solve this problem.

Abhiram824, 2025-05-04:
Hi, there could be a couple of reasons for this issue:

1. Your `MUJOCO_EGL_DEVICE_ID` environment variable is also set. Robosuite defaults to using this variable when it is set. See [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/renderers/context/egl_context.py#L39)
2. The `self.device_id` variable in the environment class is being set somewhere. Robosuite also uses this variable in deciding which device to use (if explicitly set).
3. Not sure if this is being done when running the script but  when you run `export CUDA_VISIBLE_DEVICES=1` make sure there are no spaces around the equal sign.

Abhiram824, 2025-06-10:
Closing this for now; feel free to open back up if you have any follow-ups.

closed 2025-06-10 by Abhiram824