ARISE-Initiative/robosuite #693: Controller_configs: control_delta or input_type

LucienJi, 2025-05-05:
### System Info

```Shell
Ubuntu with robosuite 1.5.1.
```

### Reproduction

`
ph_data_path = "some_data.hdf5"

env_meta = FileUtils.get_env_metadata_from_dataset(dataset_path=ph_data_path)

env_meta['env_kwargs']['controller_configs']['body_parts']['right']['control_delta'] = False

env = EnvUtils.create_env_for_data_processing(
        env_meta=env_meta,
        camera_names=['agentview'],
        camera_height=128,
        camera_width=128,
        reward_shaping=False,
            render=False,
            render_offscreen=True,
            use_image_obs=True,
            use_depth_obs=False,
        )
`

However, if we look at 'robosuite/robosuite/controllers/parts/arm/osc.py', the input_type is still "delta" instead of "absolute". 

Basically, we need to use "input_type" instead of "control_delta". While the metadata in the offline dataset still use the "control_delta".

Abhiram824, 2025-05-08:
Hello, could you share which version of robosuite the dataset you collected is using? Older versions of robosuite used the `control_delta` to determine action space and the newer versions use `input_type`

Abhiram824, 2025-06-18:
Closing for now. Feel free to open back up if you have any follow-up questions.

closed 2025-06-18 by Abhiram824