ARISE-Initiative/robosuite #694: unrecognized plugin 'mujoco.sensor.touch_grid'

gokceay, 2025-05-09:
### System Info

```Shell
robosuite version: 1.5.1
OS: Ubuntu 24.04.2 LTS

When I added touch_grid sensors on robotiq_gripper_85.xml mujoco able to open it without problem but when I tried to simulate with robosuite it crashes.
```

### Information

the error is 

"Traceback (most recent call last):
  File "/media/gokce-guven/Phoebe10TB/cobot/robosuite/test.py", line 29, in <module>
    model = world.get_model(mode="mujoco")
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/gokce-guven/Phoebe10TB/cobot/robosuite/robosuite/models/base.py", line 145, in get_model
    model = mujoco.MjModel.from_xml_string(string.getvalue())
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: Error: unrecognized plugin 'mujoco.sensor.touch_grid'
Element name 'gripper0_left_touch_grid', id 2, line 220
"

### Reproduction

the code is 
"import os
import mujoco
from robosuite.models import MujocoWorldBase
from robosuite.models.arenas import TableArena
from robosuite.models.robots import Mycobot
from robosuite.models.grippers import gripper_factory
from robosuite.models.objects import BallObject
from mujoco.viewer import launch

world = MujocoWorldBase()

robot   = Mycobot()
gripper = gripper_factory('Robotiq85Gripper')
robot.add_gripper(gripper)
robot.set_base_xpos([0.0, 0, 0])
world.merge(robot)

arena = TableArena()
arena.set_origin([0.8, 0, 0])
world.merge(arena)

sphere = BallObject(name="sphere", size=[0.04], rgba=[0,0.5,0.5,1]).get_obj()
sphere.set('pos', '1.0 0 1.0')
world.worldbody.append(sphere)

# Now your XML’s <plugin plugin="mujoco.sensor.touch_grid"> blocks will be recognized
model = world.get_model(mode="mujoco")
data  = mujoco.MjData(model)

viewer = launch(model, data)
while data.time < 1:
    mujoco.mj_step(model, data)"

kevin-thankyou-lin, 2025-05-12:
Are you able to load the new xml directly in mujoco?

i.e. try `python -m mujoco.viewer --mjcf=<new_xml>` and see if that works?

gokceay, 2025-05-13:
> Are you able to load the new xml directly in mujoco?
> 
> i.e. try `python -m mujoco.viewer --mjcf=<new_xml>` and see if that works?

Yes the model can be load directly with the command  python -m mujoco.viewer --mjcf=<new_xml> but not with robosuite

kevin-thankyou-lin, 2025-05-29:
> do you have any suggestion for this error? How can I fix this?
> Thanks in advance

Thanks for your patience - I suspect the issue is that robosuite isn’t
handling plugins correctly (maybe these are being discarded). Maybe look
into the (base) env classes, I can take a look later.

abhihjoshi, 2025-06-06:
Quick follow up on this, can you export the xml and see if the plugin is added? Afaik Robosuite at the moment does not merge plugins when merging all the models to create the scene. This is however a really nice feature we should definitely add.

gokceay, 2025-06-06:
> Quick follow up on this, can you export the scene xml and see if the plugin is added? Afaik Robosuite at the moment does not merge plugins when merging all the models to create the scene. This is however a really nice feature we should definitely add.

What do you mean by export the scene xml? For the plugin I added touch grid sensor plugin at the top of the gripper file like this:

<mujoco model="robotiq_gripper_85_model">
<compiler/>
<extension>
<plugin plugin="mujoco.sensor.touch_grid"/>
</extension>
<asset> 

and added the touch grid at the end of the file

<plugin name="left_touch_grid" plugin="mujoco.sensor.touch_grid" objtype="site" objname="left_finger_ft_mount" >
          <config key="size"     value="4 4"/>
          <config key="fov"      value="90 60"/>
          <config key="gamma"    value="0"/>
          <config key="nchannel" value="1"/>
        </plugin>
        <plugin name="right_touch_grid" plugin="mujoco.sensor.touch_grid" objtype="site" objname="right_finger_ft_mount" >
          <config key="size"     value="4 4"/>
          <config key="fov"      value="90 60"/>
          <config key="gamma"    value="0"/>
          <config key="nchannel" value="1"/>
        </plugin >

abhihjoshi, 2025-06-06:
When creating a robosuite environment, we first have to merge all the different models and their respective mjcf files together. When merging, we look for specific tags (see [here](https://github.com/ARISE-Initiative/robosuite/blob/77a4751233c29456a5381209e30dd0dbf39a6557/robosuite/models/base.py#L83)) and merge accordingly. However, at the moment, we do not merge the plugin tag from mujoco. As a result, even if your gripper has the plugin tag added for this, it won't get added to the final scene.

gokceay, 2025-06-06:
Should I change /robosuite/models/base.py file and ? By the scene xml do you mean arena xml files for the chosen environment? By the way thanks a lot for your help!

gokceay, 2025-06-06:
> By scene xml, I mean the "final" xml loaded when you say mujoco.load_from_xml (or something similar, I forgot exactly what the function name is). We can maybe make a feature request for this.

I opened a feature request https://github.com/ARISE-Initiative/robosuite/issues/715

gokceay, 2025-06-06:
> By scene xml, I mean the "final" xml loaded when you say mujoco.load_from_xml (or something similar, I forgot exactly what the function name is). We can maybe make a feature request for this.

I would like to ask another question. I have custom Mycobot robot env in robosuite and I am training it with robomimic ACT model. I would like to train with tactile information. Do you have any examples or tutorials for this?

Thanks in advance