ARISE-Initiative/robosuite #695: Mujoco xml file for Figure GR1?

forresti, 2025-05-12:
One of the robots in Robosuite is the Figure GR1. So, I was thinking that this repo would include a Mujoco xml file for the GR1. But, I haven't been able to find it.

What I'm trying to understand is:
1. How is the GR1 robot specified in the code? 
2. Is there a Mujoco xml file?
3. Are there meshes and textures for the GR1?

Thanks!

Abhiram824, 2025-05-12:
1. The file defines the controller config/defines relevant parameters/flags needed to control the GR1 robot. It does not contain the actual joint definitions. The joints are defined in the mujoco xml file [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/robots/gr1/robot.xml)
2. The xml can be found [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/robots/gr1/robot.xml)
3. The meshes /textures can be found in this [folder](https://github.com/ARISE-Initiative/robosuite/tree/master/robosuite/models/assets/robots/gr1).

closed 2025-05-12 by forresti

forresti, 2025-05-13:
@Abhiram824, I have a follow-up question. I opened [the xml file](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/robots/gr1/robot.xml) in Mujoco, and I notice the hands are missing. Are the hands defined somewhere else?

[image]

forresti, 2025-05-13:
Aha, I think I figured it out. 

- https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/grippers/inspire_left_hand.xml
- https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/grippers/inspire_right_hand.xml
- https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/grippers/fourier_left_hand.xml
- https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/grippers/fourier_right_hand.xml