ARISE-Initiative/robosuite #704: Unable to control mobile base in device control demo

jarkenau, 2025-05-19:
Hi, I’m trying to use Robosuite for a project involving mobile manipulation with a platform like the Tiago. However, I can’t figure out how to control the robot’s base.

I expected this to work in the [device control demo](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/demos/demo_device_control.py) when selecting the Tiago robot. From what I understand, the Tiago is modeled with a `NullMobileBase`: https://github.com/ARISE-Initiative/robosuite/blob/77a4751233c29456a5381209e30dd0dbf39a6557/robosuite/models/robots/manipulators/tiago_robot.py#L17-L22 which appears to define a holonomic base: https://github.com/ARISE-Initiative/robosuite/blob/77a4751233c29456a5381209e30dd0dbf39a6557/robosuite/models/assets/bases/null_mobile_base.xml#L1-L19

When running the demo, pressing `b` to switch control modes and then using the arrow keys does nothing. Is mobile base control not supported in this context, or am I missing a configuration step?

I’m running the latest master branch and installed with `pip install .` Any help is appreciated!

Abhiram824, 2025-05-19:
Hi, it may be an issue with the `demo_device_control` script. Can you try running the following and let me know if you are able to perform mobile base control: `python robosuite/scripts/collect_human_demonstrations.py  --robots Tiago`

jarkenau, 2025-05-20:
I ran `mjpython robosuite/scripts/collect_human_demonstrations.py --robots Tiago` on macOS, and I can confirm that base control works properly there.

jarkenau, 2025-05-23:
I'm not sure how this is connected, but if you remove the `DataCollectionWrapper` in `collect_human_demonstrations.py`, the robot base stops moving. Here’s the relevant code snippet: https://github.com/ARISE-Initiative/robosuite/blob/77a4751233c29456a5381209e30dd0dbf39a6557/robosuite/scripts/collect_human_demonstrations.py#L325-L326