ARISE-Initiative/robosuite #705: About custom robot integration

gokceay, 2025-05-20:
I integrate my robot model Mycobot inside robosuite, by following the tutorial : https://docs.google.com/document/d/1bSUKkpjmbKqWyV5Oc7_4VL4FGKAQZx8aWm_nvlmTVmE/edit?tab=t.0

and then I have changed the files as defined below

Changed files
Mycobot is added to the list
/robosuite/robosuite/utils/input_utils.py
robots = {"Sawyer", "Panda", "Jaco", "Kinova3", "IIWA", "UR5e", "SpotWithArmFloating", "XArm7", "Mycobot"}

To register the robot in robosuite:
/robosuite/robosuite/models/robots/manipulators/__init__.py
Line: from .mycobot_robot import Mycobot
Added to the file

Sensors and skin mounting zones are added to gripper:
/robosuite/robosuite/models/robots/manipulators/robotiq_gripper_85.xml

For robot class mapping mycobot added as a fixedbaserobot /robosuite/robosuite/robots/__init__.py

Added folders
/robosuite/robosuite/models/assets/robots/mycobot
This tutorial is followed:
https://docs.google.com/document/d/1bSUKkpjmbKqWyV5Oc7_4VL4FGKAQZx8aWm_nvlmTVmE/edit?tab=t.0 

Added files
/robosuite/robosuite/models/robots/manipulators/mycobot_robot.py is added and Robotiq85Gripper defined as default gripper.

/robosuite/robosuite/controllers/config/robots/default_mycobot.json

Then I created human demonstration data from robosuite and then converted it to robomimic dataset . I used the code below for training:

https://colab.research.google.com/github/AssistiveRoboticsUNH/bc_tutorial/blob/main/robomimic_tasks/train_lift_minimal.ipynb

Eventhough I trained successfully with 280 demonstrations and the loss value was decreasing none of my rollouts was successfull.

What could be the reason for this?

Also mycobot gives warnings as below

[robosuite WARNING] The config has defined for the controller "left", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for left from self.part_controller_config. (robot.py:151)
WARNING:robosuite_logs:The config has defined for the controller "left", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for left from self.part_controller_config.
[robosuite WARNING] The config has defined for the controller "torso", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for torso from self.part_controller_config. (robot.py:151)
WARNING:robosuite_logs:The config has defined for the controller "torso", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for torso from self.part_controller_config.
[robosuite WARNING] The config has defined for the controller "head", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for head from self.part_controller_config. (robot.py:151)
WARNING:robosuite_logs:The config has defined for the controller "head", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for head from self.part_controller_config.
[robosuite WARNING] The config has defined for the controller "base", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for base from self.part_controller_config. (robot.py:151)
WARNING:robosuite_logs:The config has defined for the controller "base", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for base from self.part_controller_config.
[robosuite WARNING] The config has defined for the controller "legs", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for legs from self.part_controller_config. (robot.py:151)
WARNING:robosuite_logs:The config has defined for the controller "legs", but the robot does not have this component. Skipping, but make sure this is intended.Removing the controller config for legs from self.part_controller_config.

Also for /robosuite/robosuite/controllers/config/robots/default_mycobot.json
this file I have used Panda's config file but they have different number of joints. When I changed the config file as for less number of joints mycobot is not working when I use panda default it is working but gives warnings and training was not successful.

Any help could be appreaciated.
Thanks in advance

Abhiram824, 2025-05-31:
If you were able to successfully collect demonstrations, then the issue probably isn't from integrating your custom robot. The warnings simply come from the fact that your robot only has arm and gripper components and is nothing to worry about. This issue might be better suited for the robomimic repo, but here are some things that might be helpful to identify your issue:

1. Inspect the rollouts and see if the robot is performing reasonable behavior. Perhaps the training is going well but needs to be run for longer
2. Try replaying actions from your collected demonstrations in the rollout environment. Perhaps there are some differences between the environment  used for data collection and the environment used for rollouts

gokceay, 2025-06-01:
Thanks a lot I solved this issue

closed 2025-06-01 by gokceay