ARISE-Initiative/robosuite #706: create_composite_robot not support bimanual robots

liyiming001, 2025-05-21:
### System Info

```Shell
Ubuntu: 20.04
robosuite: 1.5.1
```

### Reproduction

I use utils.robot_composition_utils.create_composite_robot function to register a compose robot with different robot 、base and griper.
It does work well in single robot with different base and griper. However, when I choose bimanual robot like GR1, tiago or else, robot can not render in Gui without any error info. Is it the same with you? 
Hope for your answer, thx! 

robosuite version: 1.5.1. Some test code is blow:
`    robots = cfg["robots"]
    griper = cfg["griper"] if cfg["griper"] != "None" else None
    base = cfg["base"] if cfg["base"] != "None" else None
    robot_mode = cfg["robot_mode"]
    # 当前只支持
    if robot_mode == "single":
        robot_name = cfg["robot_name"]
        # register composite robot
        cu.create_composite_robot(robot_name, robots, base, griper)
    elif robot_mode == "bimanual":
        robot_name = robots
    else:
        print(f"not support current type: {robot_mode}")
        exit()

    # 3.Get controller config
    arm_controller_config = suite.load_part_controller_config(default_controller=cfg["controller"])
    arm_controller_config["input_ref_frame"] = "world"
    arm_controller_config["input_type"] = "absolute"

    controller_config = refactor_composite_controller_config(
        arm_controller_config, robot_name, ["right", "left"]
    )

    # controller_config = load_composite_controller_config(controller="BASIC", robot=robot_name)

    # Create argument configuration
    config = {
        "env_name": cfg["environment"],
        "robots": robot_name,
        "controller_configs": controller_config,
    }`

Abhiram824, 2025-05-31:
Hi, I was able to load bimanual robots without any issues with the following command `python robosuite/demos/demo_composite_robot.py --robot Tiago --grippers PandaGripper` can you try this and let me know if you have any issues

Abhiram824, 2025-07-09:
Closing for now, feel free to open back up if any other issues come up.

closed 2025-07-09 by Abhiram824