ARISE-Initiative/robosuite #707: OSC_Pose delta action format, how to obtain from absolute EE poses?

gtziafas, 2025-05-22:
Hey there! I have a basic question about the `OSC_POSE` controller, which I cannot seem to identify in the documentation. 

I'm using the two-arm Panda Franka environments, with controller config like this:
```
{
    "type": "BASIC",
    "body_parts": {
        "arms": {
            "right": {
                "type": "OSC_POSE",
                "input_max": 1,
                "input_min": -1,
                "output_max": [0.05, 0.05, 0.05, 0.5, 0.5, 0.5],
                "output_min": [-0.05, -0.05, -0.05, -0.5, -0.5, -0.5],
                "kp": 150,
                "damping_ratio": 1,
                "impedance_mode": "fixed",
                "kp_limits": [0, 300],
                "damping_ratio_limits": [0, 10],
                "position_limits": null,
                "orientation_limits": null,
                "uncouple_pos_ori": true,
                "control_delta": true,
                "input_type": "delta",
                "input_ref_frame": "base",
                "interpolation": null,
                "ramp_ratio": 0.2,
                "gripper": {
                    "type": "GRIP"
                }
            }
        }
    }
}
```

From the documentation, I see that the action vector expected by the controller is 14-dim (right delta pos, right delta axis-angles, right gripper, left delta pos, left delta axis-angles, left gripper). 

My question is, if I have a policy that produces absolute EE poses (e.g. wrt. world frame, or I can express them in each robot's base frame) how do I appropriately convert them to the delta format that is expected by the controller?

Intuitively, if I have a `curr_pose` and `next_pose` for each robot arm, I would compute the delta like this:
```python
curr_pose_base = transform_from_world_to_base(curr_pose)
next_pose_base = transform_from_world_to_base(next_pose)

delta_pos = next_pose_base[:3,3] - curr_pose_base[:3,3]
delta_rot = curr_pose_base[:3,:3].T @ next_pose_base[:3,:3]
delta_axis_angle = mat2axangle(delta_rot)

# scale to expected range
delta_pos_norm = delta_pos  / output_max_pos
delta_rot_norm  = delta_axis_angle / output_max_rot

delta_action = np.concatenate([delta_pos_norm, delta_rot_norm])
```

where `output_max` are read from the controller config ( `0.05` for pos and `0.5` for orn).

However, this doesnt seem to work. I have some demonstration data (with delta actions that work), from which I also read the trajectory of absolute EE poses. When I apply the above, I get sth like:
```
Computed delta from absolute EE: [-0.00124251, -0.00686029, -0.00860876, 1.81888703e-03, -6.20935713e-05, -1.36323184e-02]
Ground-truth delta from dataset: [-0.05596172  0.07625895  0.03767472 -0.00120852 -0.01216589 -0.08203301]
```
you see that they are orders of magnitude different. 

It would be greatly appreciated if you could provide some context (or a pointer to an explanation / example) that demonstrates how to go from absolute EE poses to delta actions, as expected by the `OSC_POSE` controller.

Abhiram824, 2025-05-22:
Hi, there is some reference code in in osc.py that converts delta to absolute action - see [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/parts/arm/osc.py#L572). One more thing that the delta actions stored in the dataset are not scaled/normalized, so you shouldn't have to apply that when trying to convert.

Hope this helps!

gtziafas, 2025-06-02:
Hey there, thanks for the pointer!

Here's a reverse (absolute to delta) implementation that works, maybe its also of interest for you to integrate etc.:

```python
def abs_to_delta_action(self, abs_action, goal_update_mode=None):
    """
    Helper function that converts absolute action into delta action.
    
    This is the inverse of delta_to_abs_action. It takes an absolute EE pose
    and converts it to a delta action that can be executed by the controller.
    
    Args:
        abs_action (np.array): Desired absolute EE pose [x, y, z, ax, ay, az]
                              where (x,y,z) is position and (ax,ay,az) is axis-angle orientation
        goal_update_mode (str): either "achieved" (current achieved position) or "desired" (current desired goal)
                               If None, uses self._goal_update_mode
    
    Returns:
        np.array: Delta action [dx, dy, dz, dax, day, daz] that can be passed to set_goal()
    """

    def _unscale_action(scaled_action):
        action_scale = abs(self.output_max - self.output_min) / abs(self.input_max - self.input_min)
        action_output_transform = (self.output_max + self.output_min) / 2.0
        action_input_transform = (self.input_max + self.input_min) / 2.0
        original_action = (scaled_action - action_output_transform) / action_scale + action_input_transform
        return original_action

    if goal_update_mode is None:
        goal_update_mode = getattr(self, '_goal_update_mode', 'achieved')
    
    assert goal_update_mode in ["achieved", "desired"], f"Invalid goal_update_mode: {goal_update_mode}"
    
    # Extract absolute position and orientation from input
    abs_pos = abs_action[0:3]
    abs_ori_axisangle = abs_action[3:6]
    abs_ori_mat = T.quat2mat(T.axisangle2quat(abs_ori_axisangle))
    
    # Determine reference position and orientation based on goal_update_mode
    if goal_update_mode == "desired":
        # Compare against current desired goal
        if self.goal_pos is None:
            # If no goal is set, initialize to current position
            if self.input_ref_frame == "base":
                ref_pos = self.world_to_origin_frame(self.ref_pos)
            elif self.input_ref_frame == "world":
                ref_pos = self.ref_pos
            else:
                raise ValueError(f"Invalid input_ref_frame: {self.input_ref_frame}")
        else:
            ref_pos = self.goal_pos
            
        if self.goal_ori is None:
            # If no goal is set, initialize to current orientation
            if self.input_ref_frame == "base":
                ref_ori_mat = self.goal_origin_to_eef_pose()[:3, :3]
            elif self.input_ref_frame == "world":
                ref_ori_mat = self.ref_ori_mat
            else:
                raise ValueError(f"Invalid input_ref_frame: {self.input_ref_frame}")
        else:
            ref_ori_mat = self.goal_ori
            
    elif goal_update_mode == "achieved":
        # Compare against current achieved position/orientation
        if self.input_ref_frame == "base":
            ref_pos = self.world_to_origin_frame(self.ref_pos)
            ref_ori_mat = self.goal_origin_to_eef_pose()[:3, :3]
        elif self.input_ref_frame == "world":
            ref_pos = self.ref_pos
            ref_ori_mat = self.ref_ori_mat
        else:
            raise ValueError(f"Invalid input_ref_frame: {self.input_ref_frame}")
    else:
        raise ValueError(f"Invalid goal_update_mode: {goal_update_mode}")
    
    # Compute position delta
    delta_pos = abs_pos - ref_pos
    
    # Compute orientation delta
    # We need to find the rotation that transforms ref_ori_mat to abs_ori_mat
    # This is: rotation_error = abs_ori_mat * ref_ori_mat.T
    rotation_error_mat = np.dot(abs_ori_mat, ref_ori_mat.T)
    rotation_error_quat = T.mat2quat(rotation_error_mat)
    delta_ori = T.quat2axisangle(rotation_error_quat)
    
    # Combine position and orientation deltas
    delta_action = np.concatenate([delta_pos, delta_ori])

    # Unscale action
    delta_action_unscaled = _unscale_action(delta_action)
    
    return delta_action_unscaled
```

Abhiram824, 2025-06-08:
Awesome, glad I could help and thanks for the code - will take a look!

closed 2025-06-08 by Abhiram824