ARISE-Initiative/robosuite #720: [Feature]: Add support for SO-ARM100

aleSuglia, 2025-06-16:
### Description

The SO‑101 is the next‑generation version of the SO‑100 robot arm, originally designed by the [RobotStudio](https://www.therobotstudio.com/) in collaboration with [Hugging Face](https://huggingface.co/lerobot). It has improved wiring, is easier to assemble (no gear removal) and uses updated motors for the leader arm.

It would be fantastic to support it as one of the arms to streamline access to better simulation engines that can then be integrated into LeRobot. 

### Suggested Solution

The SO-101 is already supported as a Mujoco model and you can find the implementation here: https://github.com/TheRobotStudio/SO-ARM100/tree/main/Simulation/SO101

I have started integrating it into the `robots` by implementing a `ManipulatorModel`, but unfortunately, when running the tests, there are some issues associated with the default classes that are used in the XML specification.

Leesangoh, 2025-07-16:
Did it work?

aleSuglia, 2025-08-26:
Sorry I haven't had the chance to check this further. I hope the authors can help wit this!