ARISE-Initiative/robosuite #725: Extracting complete models?

gpu-daal, 2025-06-18:
Hi, that's a very cool project and I'm very impressed by the level of details! 
I was just wondering if it's possible to extract the complete mjcf models as a single xml file. Right now extracting GR1 model only extracts the robot with no grippers attached, so I assume assemble of the model is done during runtime, but I want to see how my model behaves in a free environment (i.e mujoco, brax, etc) with lots of joints, so I'd like to have Inspire grippers attached.

Thank you!

gpu-daal, 2025-06-19:
Oh, that's actually very helpful, thank you! How did you guys make your humanoid stay still? Is it a part of the training, or did you hardcode/disable some joints?

Abhiram824, 2025-06-19:
The joints are removed - see [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/robots/manipulators/gr1_robot.py#L107)

Abhiram824, 2025-07-17:
Closing for now; feel free to open back up if you have any follow up questions

closed 2025-07-17 by Abhiram824