ARISE-Initiative/robosuite #735: Error message running Sawyer in Example script

Mi-Przystupa, 2025-07-17:
### System Info

```Shell
ran Pip install robosuite

robosuite==1.5.1
```

### Information

I tried running the demo script for the gym wrapper ([link](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/demos/demo_gym_functionality.py)) with version 1.5.1 of robosuite and it gave the below error:

```
Exception has occurred: ValueError
Error: mesh volume is too small: robot0_head_1 . Try setting inertia to shell
Element name 'robot0_head_1', id 3, line 32
  File "/home/mprzystupa/git/manipulation_rl_benchmarking/maniskill_ppo/robosuite_examples/gym_test.py", line 52, in <module>
    suite.make(
ValueError: Error: mesh volume is too small: robot0_head_1 . Try setting inertia to shell
Element name 'robot0_head_1', id 3, line 32
```

I tried some modifications to the config, but that didn't seem to change anything:

```
import robosuite as suite
from robosuite.wrappers import GymWrapper
from robosuite.controllers.composite.composite_controller_factory import refactor_composite_controller_config

if __name__ == "__main__":
    config = suite.load_part_controller_config(default_controller="OSC_POSE")
            #any limbs you specify are added to the configuration, this supresses warnings about missing limbs
    config = refactor_composite_controller_config(config, "Sawyer", arms = ['right']) 
    # Notice how the environment is wrapped by the wrapper

    config = { 
            "env_name": "Lift", # the name of the environment 
            "robots": "Sawyer",  # use Sawyer robot
            "use_camera_obs": True,  # do not use pixel observations
            "has_offscreen_renderer": True,  # not needed since not using pixel obs
            "has_renderer": True,  # make sure we can render to the screen
            "reward_shaping": True,  # use dense rewards
            "control_freq": 20,  # control should happen fast enough so that simulation looks smooth
            "controller_configs": config,
        }
    env = GymWrapper(
        suite.make(
            **config
                    )
```

### Reproduction

Steps to reproduce:
1. Create a python=3.10 environment with conda
2. pip install robosuite (pip install robosuite==1.5.1 to be more specific)
3. Try to run the script in the [link](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/demos/demo_gym_functionality.py) with the Sawyer

### Expected behavior

The Sawyer robot should not be throwing this error

Abhiram824, 2025-07-17:
Hi could you share what mujoco version you have? could you try using `mujoco==3.3.1`

Mi-Przystupa, 2025-07-17:
I was using mujoco==3.3.4

I downgraded to mujoco==3.3.1 and received the same error

If it matters, I installed maniskill before robosuite if that helps:

```
pip install --upgrade mani_skill
```

Abhiram824, 2025-08-01:
Closing this for the time being, feel free to reopen if you have any follow up problems

closed 2025-08-01 by Abhiram824

ParsaJafarian, 2025-08-31:
> Ah I think I know what the issue may be. Could you try either downgrading mujoco to 3.2.3 Or (if you need the specific mujoco version) install robosuite from source.

I confirm that downgrading mujoco to 3.2.3 fixed the issue for me.