ARISE-Initiative/robosuite #752: Illegal instruction (core dumped)

xiaoc10020015, 2025-09-16:
### System Info

```Shell
def _check_sensor_validity(self):
        """
        Internal function that checks the validity of this observable's sensor. It does the following:

            - Asserts that the inputted sensor has its __modality__ attribute defined from the sensor decorator
            - Asserts that the inputted sensor can handle the empty dict {} arg case
            - Updates the corresponding name, and data-types for this sensor
        """
        try:
            _ = self.modality
            if '_site' not in self._sensor.__name__:
                self._data_shape = np.array(self._sensor({})).shape
                self._is_number = len(self._data_shape) == 1 and self._data_shape[0] == 1
```

### Reproduction

在 程序启动就会崩溃

Abhiram824, 2025-09-17:
Hi, what script are you running to get this error?

xiaoc10020015, 2025-09-17:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，您的来信本人已收到，将尽快给您回复。

Abhiram824, 2025-10-07:
Closing this issue for now, feel free to reopen if you have any follow-up issues

closed 2025-10-07 by Abhiram824